Linutronix/elbe #233: Stretch initvm build fails on buster

AndreySV, 2019-08-20:
command ```elbe initvm create elbe-init-big-machine.xml``` fails at step 14 'Select and install software'. According to syslog d-i complains that elbe-* packages are unauthenticated, because of missing public key for elbe repositories.

Host system:
Debian Buster (amd64),
ELBE 10 (from linuxtronix debian repo)

I've used ```<noauth>``` tag as a workaround for this problem.

bgermann, 2019-08-21:
I cannot reproduce the issue. The public key is referenced in the file and the packages are signed. Please try again. Maybe it was just a connection issue on your side.

AndreySV, 2019-08-21:
I tried three times and build fails always with the same error. I'm expecting default configuration to work and the problem is probably on my side.

Keyring is available on file system:
```# ls -l /usr/share/keyrings/
lrwxrwxrwx    1 root     root            26 Aug 21 08:18 archive.gpg -> debian-archive-keyring.gpg
-rw-r--r--    1 root     root         36941 May 25  2017 debian-archive-keyring.gpg
-rw-r--r--    1 1001     1001      33161342 Aug 21 08:17 elbe-keyring.gpg```

```
# md5sum /usr/share/keyrings/*
7b3f0258ee06be36641ddb945a7aa1e1  /usr/share/keyrings/archive.gpg
7b3f0258ee06be36641ddb945a7aa1e1  /usr/share/keyrings/debian-archive-keyring.gpg
002ec45c009c21d1d47dbea566f9d771  /usr/share/keyrings/elbe-keyring.gpg
```

In installer syslog I see following messages:
```
...
Aug 21 08:22:25 in-target: Get:1 http://debian.linutronix.de/elbe stretch InRelease [41.2 kB]
Aug 21 08:22:25 in-target: Ign:1 http://debian.linutronix.de/elbe stretch InRelease
Aug 21 08:22:25 in-target: Get:2 http://debian.linutronix.de/elbe stretch/main amd64 Packages [59.3 kB]
Aug 21 08:22:25 in-target: Fetched 101 kB in 0s (226 kB/s)
Aug 21 08:22:25 in-target: Reading package lists...
Aug 21 08:22:25 in-target: 
Aug 21 08:22:25 in-target: W: GPG error: http://debian.linutronix.de/elbe stretey is not available: NO_PUBKEY 36AA35FF22BB8F84 be verified because the
Aug 21 08:22:25 in-target: W: The repository 'http://debian.linutronix.de/elbe stretch InRelease' is not signed.
Aug 21 08:22:26 in-target: Get:1 http://debian.linutronix.de/elbe stretch InRelease [41.2 kB]
Aug 21 08:22:26 in-target: Ign:1 http://debian.linutronix.de/elbe stretch InRelease
Aug 21 08:22:25 in-target: Fetched 101 kB in 0s (226 kB/s)
Aug 21 08:22:25 in-target: Reading package lists...
Aug 21 08:22:25 in-target: 
Aug 21 08:22:25 in-target: W: GPG error: http://debian.linutronix.de/elbe stretey is not available: NO_PUBKEY 36AA35FF22BB8F84 be verified because the
Aug 21 08:22:25 in-target: W: The repository 'http://debian.linutronix.de/elbe stretch InRelease' is not signed.
Aug 21 08:22:26 in-target: Get:1 http://debian.linutronix.de/elbe stretch InRelease [41.2 kB]
Aug 21 08:22:26 in-target: Ign:1 http://debian.linutronix.de/elbe stretch InRelease
Aug 21 08:22:26 in-target: Get:2 http://debian.linutronix.de/elbe stretch/main S--More-- 
ources [10.3 kB]
Aug 21 08:22:26 in-target: Fetched 51.5 kB in 0s (168 kB/s)
Aug 21 08:22:26 in-target: Reading package lists...
Aug 21 08:22:26 in-target: 
Aug 21 08:22:26 in-target: W: GPG error: http://debian.linutronix.de/elbe stretey is not available: NO_PUBKEY 36AA35FF22BB8F84 be verified because the
Aug 21 08:22:26 in-target: W: The repository 'http://debian.linutronix.de/elbe stretch InRelease' is not signed.
Aug 21 08:22:26 in-target: Get:1 http://debian.linutronix.de/elbe-common stretch InRelease [41.2 kB]
Aug 21 08:22:27 in-target: Ign:1 http://debian.linutronix.de/elbe-common stretch InRelease
Aug 21 08:22:27 in-target: Get:2 http://debian.linutronix.de/elbe-common stretch/main amd64 Packages [10.6 kB]
Aug 21 08:22:27 in-target: Fetched 51.8 kB in 0s (187 kB/s)
Aug 21 08:22:27 in-target: Reading package lists...
Aug 21 08:22:27 in-target: 
Aug 21 08:22:27 in-target: W: GPG error: http://debian.linutronix.de/elbe-commoublic key is not available: NO_PUBKEY 36AA35FF22BB8F84 be verified becu
Aug 21 08:22:26 in-target: W: The repository 'http://debian.linutronix.de/elbe stretch InRelease' is not signed.
Aug 21 08:22:26 in-target: Get:1 http://debian.linutronix.de/elbe-common stretch InRelease [41.2 kB]
Aug 21 08:22:27 in-target: Ign:1 http://debian.linutronix.de/elbe-common stretch InRelease
Aug 21 08:22:27 in-target: Get:2 http://debian.linutronix.de/elbe-common stretch/main amd64 Packages [10.6 kB]
Aug 21 08:22:27 in-target: Fetched 51.8 kB in 0s (187 kB/s)
Aug 21 08:22:27 in-target: Reading package lists...
Aug 21 08:22:27 in-target: 
Aug 21 08:22:27 in-target: W: GPG error: http://debian.linutronix.de/elbe-commoublic key is not available: NO_PUBKEY 36AA35FF22BB8F84 be verified becu
Aug 21 08:22:27 in-target: W: The repository 'http://debian.linutronix.de/elbe-c--More-- 
ommon stretch InRelease' is not signed.
Aug 21 08:22:27 in-target: Get:1 http://debian.linutronix.de/elbe-common stretch InRelease [41.2 kB]
Aug 21 08:22:27 in-target: Ign:1 http://debian.linutronix.de/elbe-common stretch InRelease
Aug 21 08:22:27 in-target: Get:2 http://debian.linutronix.de/elbe-common stretch/main Sources [19.1 kB]
Aug 21 08:22:27 in-target: Fetched 60.3 kB in 0s (184 kB/s)
Aug 21 08:22:27 in-target: Reading package lists...
Aug 21 08:22:27 in-target: 
Aug 21 08:22:27 in-target: W: GPG error: http://debian.linutronix.de/elbe-commoublic key is not available: NO_PUBKEY 36AA35FF22BB8F84 be verified becu
Aug 21 08:22:27 in-target: W: The repository 'http://debian.linutronix.de/elbe-common stretch InRelease' is not signed.
Aug 21 08:22:28 in-target: Get:1 http://ftp.de.debian.org/debian stretch-backports InRelease [91.8 kB]
Aug 21 08:22:28 in-target: Get:2 http://ftp.de.debian.org/debian stretch-backports/main amd64 Packages [603 kB]
Aug 21 08:22:28 in-target: Get:3 http://ftp.de.debian.org/debian stretch-backports/main Translation-en [461 kB]
Aug 21 08:22:29 in-target: Fetched 1,156 kB in 1s (900 kB/s)
...
Aug 21 08:23:19 in-target:   ucf x11-common x11-xserver-utils xml-core xz-utils
Aug 21 08:23:19 in-target: 0 upgraded, 159 newly installed, 0 to remove and 0 not upgraded.
[... cut ...]
Aug 21 08:23:19 in-target: WARNING: The following packages cannot be authenticated!
Aug 21 08:23:19 in-target:   elbe-schema python-elbe-bin python-elbe-common python-elbe-buildenv
Aug 21 08:23:19 in-target:   elbe-daemon elbe-soap debathena-transform-lighttpd
d without --allow-unauthenticatedre were unauthenticated packages and -y was use--More-- 
Aug 21 08:23:20 main-menu[2765]: WARNING **: Configuring 'pkgsel' failed with error code 100
Aug 21 08:23:20 main-menu[2765]: WARNING **: Menu item 'pkgsel' failed.
...
```

bgermann, 2019-08-21:
You have a "stretey" (not stretch) repo defined. Maybe this causes the issue.

bgermann, 2019-08-21:
Ah, I see. There are just some characters missing, right?

AndreySV, 2019-08-21:
I'm using default configuration without any changes at all. These are artifact because of copying from installer console. I've checked syslog with nano extra for this. Messages contains 'stretch', not 'stretey'. 

Do you know any way to copy logs from installer?

AndreySV, 2019-08-21:
I've checked contents of elbe-keyring.gpg in initrd for missing key (36AA35FF22BB8F84).
```
$ gpg2 --no-default-keyring --keyring elbe-keyring.gpg --list-keys | grep 36AA

```
In my system it's available on trusted keyring.
```
$ gpg2 --no-default-keyring --keyring /etc/apt/trusted.gpg --list-keys | grep 36AA
gpg: please do a --check-trustdb
pub   rsa2048/0x36AA35FF22BB8F84 2012-11-23 [SC] [expires: 2022-12-10]
      Key fingerprint = 10E0 A535 5609 F7C6 A55A  17F6 36AA 35FF 22BB 8F84

```

bgermann, 2019-08-21:
What is your host system? Maybe, if it is too old it does not support the hashing/signing algorithms that are used.

AndreySV, 2019-08-21:
Host system is debian Buster. I'm still investigating why elbe-keyring doesn't contain any actual ELBE keys.

bgermann, 2019-08-23:
Okay, I can reproduce the issue now. Thanks for the patch!

AndreySV, 2019-08-23:
Thanks for merging.

commit 7f842955 (2019-08-28): debian changelog: bump version to v11

  * move <src-opts> into <src-cdrom> node
  * Update Debian codenames (buster is stable now)
  * command init: don't use user's gnupg configuration file (Fixes: #233)
  * grubinstaller: enable UEFI (secure) boot with shim
  * fix grubinstall, when /var is separate partition (Fixes: #230)
commit d13bb8a4 (2019-08-28): command init: don't use user's gnupg configuration file

if user's gnupg configuration file contained options to include
other keyrings, then 'elbe-keyring' wasn't created at all

gpg error message if gnupg conf has keyring option:
gpg: keyblock resource
'.../elbe/initvm/.elbe-in/elbe-keyring':
No such file or directory
gpg: key 0x36AA35FF22BB8F84: 1 signature not checked due to a missing key
gpg: no writable keyring found: Not found
gpg: error reading '[stdin]': General error
gpg: import from '[stdin]' failed: General error
gpg: Total number processed: 0

Because of that resulting 'elbe-keyri [... cut]
closed 2019-08-23 by AndreySV