Linutronix/elbe #253: elbe fails on signature check using Ubuntu 19.10

coveritytest, 2020-01-24:
sudo ./elbe initvm create ./examples/armhf-ti-beaglebone-black.xml

Import ubuntu-keyring-2012-archive.gpg: 
gpg: key 3B4FE6ACC0B21F32: 3 signatures not checked due to missing keys
gpg: key 3B4FE6ACC0B21F32: public key "Ubuntu Archive Automatic Signing Key (2012) <ftpmaster@ubuntu.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
Import ubuntu-keyring-2018-archive.gpg: 
gpg: key 871920D1991BC93C: 1 signature not checked due to a missing key
gpg: key 871920D1991BC93C: public key "Ubuntu Archive Automatic Signing Key (2018) <ftpmaster@ubuntu.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
Import ubuntu-keyring-2012-cdimage.gpg: 
gpg: key D94AA3F0EFE21092: 3 signatures not checked due to missing keys
gpg: key D94AA3F0EFE21092: public key "Ubuntu CD Image Automatic Signing Key (2012) <cdimage@ubuntu.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
Import microsoft.gpg: 
gpg: key EB3E94ADBE1229CF: public key "Microsoft (Release signing) <gpgsecurity@microsoft.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
--2020-01-24 11:59:32--  http://ftp.de.debian.org//debian/dists/buster/Release
Resolving ftp.de.debian.org (ftp.de.debian.org)... 141.76.2.4
Connecting to ftp.de.debian.org (ftp.de.debian.org)|141.76.2.4|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 119065 (116K)
Saving to: ‘/tmp/tmpLU149G/Release’

/tmp/tmpLU149G/Release                             100%[================================================================================================================>] 116,27K  --.-KB/s    in 0,07s   

2020-01-24 11:59:32 (1,55 MB/s) - ‘/tmp/tmpLU149G/Release’ saved [119065/119065]

Debian Archive Automatic Signing Key <ftpmaster@debian.org> (16E90B3FDF65EDE3AA7F323C04EE7237B7D453EC):
VALID (Untrusted).
Signature with unknown key: 0146DC6D4A0B2914BDED34DB648ACFD622F3D138
Signature with unknown key: 6D33866EDD8FFA41C0143AEDDCC9EFBF77E11517
[ERROR] Failure to download kernel/initrd debian Package:
[ERROR] 
[ERROR] InvalidSignature Failed to verify Release file
[ERROR] 
[ERROR] Check Mirror configuration
'elbe init' Failed
Giving up

I'm on branch releases/v12.2. I did import the keys:

sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 8B48AD6246925553
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 4dfab270caa96dfa
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys e0b11894f66aec98
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 7638d0442b90d010

bgermann, 2020-01-24:
Please install debian-archive-keyring. If you just want to use Elbe and do not want to develop you should follow the Debian buster installation described at https://elbe-rfs.org/download/

coveritytest, 2020-01-24:
Followed the instructions but the error is still there:

elbe initvm create elbe/examples/armhf-ti-beaglebone-black.xml
Import ubuntu-keyring-2012-archive.gpg: 
gpg: key 3B4FE6ACC0B21F32: 3 signatures not checked due to missing keys
gpg: key 3B4FE6ACC0B21F32: public key "Ubuntu Archive Automatic Signing Key (2012) <ftpmaster@ubuntu.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
Import ubuntu-keyring-2018-archive.gpg: 
gpg: key 871920D1991BC93C: 1 signature not checked due to a missing key
gpg: key 871920D1991BC93C: public key "Ubuntu Archive Automatic Signing Key (2018) <ftpmaster@ubuntu.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
Import ubuntu-keyring-2012-cdimage.gpg: 
gpg: key D94AA3F0EFE21092: 3 signatures not checked due to missing keys
gpg: key D94AA3F0EFE21092: public key "Ubuntu CD Image Automatic Signing Key (2012) <cdimage@ubuntu.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
Import microsoft.gpg: 
gpg: key EB3E94ADBE1229CF: public key "Microsoft (Release signing) <gpgsecurity@microsoft.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
--2020-01-24 12:34:20--  http://ftp.de.debian.org//debian/dists/buster/Release
Resolving ftp.de.debian.org (ftp.de.debian.org)... 141.76.2.4
Connecting to ftp.de.debian.org (ftp.de.debian.org)|141.76.2.4|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 119065 (116K)
Saving to: ‘/tmp/tmppCWOSn/Release’

/tmp/tmppCWOSn/Release                             100%[================================================================================================================>] 116,27K  --.-KB/s    in 0,09s   

2020-01-24 12:34:20 (1,23 MB/s) - ‘/tmp/tmppCWOSn/Release’ saved [119065/119065]

Debian Archive Automatic Signing Key <ftpmaster@debian.org> (16E90B3FDF65EDE3AA7F323C04EE7237B7D453EC):
VALID (Untrusted).
Signature with unknown key: 0146DC6D4A0B2914BDED34DB648ACFD622F3D138
Signature with unknown key: 6D33866EDD8FFA41C0143AEDDCC9EFBF77E11517
[ERROR] Failure to download kernel/initrd debian Package:
[ERROR] 
[ERROR] InvalidSignature Failed to verify Release file
[ERROR] 
[ERROR] Check Mirror configuration
'elbe init' Failed
Giving up

coveritytest, 2020-01-24:
Okay, turns out the Ubuntu package for debian-archive-keyring installs the keys only to /usr/share/keyrings but not to /etc/apt/trusted.gpg.d.

closed 2020-01-24 by coveritytest