Linutronix/elbe #254: Failed to set up loop device: Resource temporarily unavailable

martin-petzold, 2020-02-01:
Hello all,

since a while all my builds fail with the error below. I am running most recent and clean Debian Buster with most recent ELBE release (12.2) on the build machine. I'm running this platform in a VM with nested virtualization. Please find attached the XML.

I am using elbe submit via gitlab-runner user. The command is run as a user added to the groups as described in the documentation (libvirt, libvirt-qemu, and kvm). I realized, that the build uses quite a lot for RAM. That's why I am currently running the VM with one core and 32GB of RAM. Disk is 64 GB with around 45GB free space.

Command: "elbe initvm --skip-build-bin --skip-build-sources --output target submit live.xml" 

[live.xml.txt](https://github.com/Linutronix/elbe/files/4142675/live.xml.txt)

--- START

7437 [CMD] Project build was not successful, current status: build_failed
7438 mount /dev/loop0 /var/cache/elbe/423ce6a5-2379-461f-866d-676c647838c1/imagemnt
7439 [CMD] cp -a "/var/cache/elbe/423ce6a5-2379-461f-866d-676c647838c1/filesystems/0/." "/var/cache/elbe/423ce6a5-2379-461f-866d-676c647838c1/imagemnt/"
7440 [CMD] umount /dev/loop0
7441 [CMD] losetup -d /dev/loop0
7442 [CMD] losetup -o537919488 --sizelimit 2147483648 /dev/loop0 "/var/cache/elbe/423ce6a5-2379-461f-866d-676c647838c1/install"
7443 losetup: /var/cache/elbe/423ce6a5-2379-461f-866d-676c647838c1/install: failed to set up loop device: Resource temporarily unavailable
7444 [CMD] mv "/var/cache/elbe/423ce6a5-2379-461f-866d-676c647838c1/filesystems/1"/* "/var/cache/elbe/423ce6a5-2379-461f-866d-676c647838c1/target/"
7445 [ERROR] Build failed
7446 Traceback (most recent call last):
7447   File "/usr/lib/python2.7/dist-packages/elbepack/asyncworker.py", line 182, in execute
7448     skip_pbuild=self.skip_pbuilder)
7449   File "/usr/lib/python2.7/dist-packages/elbepack/elbeproject.py", line 614, in build
7450     self.targetfs.part_target(self.builddir, grub_version, grub_fw_type)
7451   File "/usr/lib/python2.7/dist-packages/elbepack/efilesystem.py", line 364, in part_target
7452     grub_fw_type)
7453   File "/usr/lib/python2.7/dist-packages/elbepack/hdimg.py", line 487, in do_hdimg
7454     grub_fw_type)
7455   File "/usr/lib/python2.7/dist-packages/elbepack/hdimg.py", line 365, in do_image_hd
7456     create_label(disk, part, ppart, fslabel, target, grub)
7457   File "/usr/lib/python2.7/dist-packages/elbepack/hdimg.py", line 278, in create_label
7458     entry.losetup("loop0")
7459   File "/usr/lib/python2.7/dist-packages/elbepack/fstab.py", line 145, in losetup
7460     (self.offset, self.size, loopdev, self.filename))
7461   File "/usr/lib/python2.7/dist-packages/elbepack/shellhelper.py", line 115, in do
7462     raise CommandError(cmd, p.returncode)

--- END

Thanks and kind regards,

Martin

oldiob, 2020-02-14:
Could you try to add:
~~~xml
<cp path="/usr/share/grub/default/grub">/etc/default/grub</cp>
~~~
**before** the *sed* commands in the **finetuning** section?

martin-petzold, 2020-02-15:
Does not help.

matlec, 2020-03-23:
Actually we run into the same issue on our build server. I was able to reproduce the behaviour by running the following command inside the `initvm` (reproducing this issue may take some attempts):
```
losetup -o2203058176 --sizelimit 524288000 /dev/loop0 "/var/cache/elbe/c42cf38f-5cf8-4cb2-bda7-37368487bc87/sda.img" && echo 1 && losetup -d /dev/loop0 && echo 2 && losetup -o2203058176 --sizelimit 524288000 /dev/loop0 "/var/cache/elbe/c42cf38f-5cf8-4cb2-bda7-37368487bc87/sda.img" && echo 3 && losetup -d /dev/loop0 && echo 4
```

Output:
```
1
2
losetup: /var/cache/elbe/c42cf38f-5cf8-4cb2-bda7-37368487bc87/sda.img: failed to set up loop device: Device or resource busy
```

What helps is putting a `sleep 1` after `losetup -d /dev/loop0`. This seems to be a race condition caused by [this line](https://github.com/torvalds/linux/blob/84df9525b0c27f3ebc2ebb1864fa62a97fdedb7d/drivers/block/loop.c#L1048) in the loop device driver. A better option than putting some random sleep in would be to poll the `losetup` output like this:
```
losetup -l /dev/loop0 -n --raw | cut -d' ' -f6
```

Would be highly appreciated to have this fix in Elbe as soon as possible since this causes our builds to fail.

oldiob, 2020-03-23:
I made a patch with your recommendation <http://elbe-devel.2373613.n4.nabble.com/elbe-devel-PATCH-fstab-Fix-issue-254-td3632.html>.  If you can test it.

matlec, 2020-03-24:
Thank you, your patch seems to have fixed our problem!

matlec, 2020-04-23:
I applied the latest patch set that you posted to the elbe-devel mailing list on April 15th on our build server. Unfortunatelly, I still face issues with your patch:
```
[INFO] 
[INFO] Archive validation
[INFO] ------------------
[INFO] 
[INFO] No Errors found
[CMD] mkdir -p /var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/filesystems
[CMD] mkdir -p /var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/imagemnt
[CMD] mkdir -p "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/filesystems/1"
[CMD] mkdir -p "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/target/"/boot/efi
[CMD] mkdir -p "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/filesystems/3"
[CMD] mkdir -p "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/target/"/mnt/root-rw
[CMD] mkdir -p "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/filesystems/2"
[CMD] mkdir -p "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/target/"/boot/grub
[CMD] mv "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/target/boot/grub"/* "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/filesystems/2"
[CMD] mkdir -p "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/filesystems/4"
[CMD] mkdir -p "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/target/"/opt
[CMD] mkdir -p "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/filesystems/0"
[CMD] mkdir -p "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/target/"/
[CMD] mv "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/target/"/* "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/filesystems/0"
[CMD] rm -f "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/sda.img"
[CMD] losetup --offset 1048576 --sizelimit 104857600 --find --show "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/sda.img"
losetup: /var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/sda.img: failed to set up loop device: Resource temporarily unavailable
[CMD] mv "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/filesystems/0"/* "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/target/"
[CMD] mv "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/filesystems/2"/* "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/target/boot/grub"
[ERROR] Build failed
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/elbepack/asyncworker.py", line 186, in execute
    skip_pbuild=self.skip_pbuilder)
  File "/usr/lib/python2.7/dist-packages/elbepack/elbeproject.py", line 627, in build
    self.targetfs.part_target(self.builddir, grub_version, grub_fw_type)
  File "/usr/lib/python2.7/dist-packages/elbepack/efilesystem.py", line 371, in part_target
    grub_fw_type)
  File "/usr/lib/python2.7/dist-packages/elbepack/hdimg.py", line 561, in do_hdimg
    grub_fw_type)
  File "/usr/lib/python2.7/dist-packages/elbepack/hdimg.py", line 439, in do_image_hd
    create_label(disk, part, ppart, fslabel, target, grub)
  File "/usr/lib/python2.7/dist-packages/elbepack/hdimg.py", line 349, in create_label
    loopdev = entry.losetup()
  File "/usr/lib/python2.7/dist-packages/elbepack/fstab.py", line 145, in losetup
    (self.offset, self.size, self.filename))
  File "/usr/lib/python2.7/dist-packages/elbepack/shellhelper.py", line 149, in get_command_out
    raise CommandError(cmd, p.returncode)
CommandError: Error: 1 returned from Command losetup --offset 1048576 --sizelimit 104857600 --find --show "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3/sda.img"
Project build was not successful, current status: build_failed
elbe control wait_busy Failed

The project will not be deleted from the initvm.
The files, that have been built, can be downloaded using:
/usr/bin/elbe control get_files --output "/elbe/scripts/../dist" "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3"

The project can then be removed using:
/usr/bin/elbe control del_project "/var/cache/elbe/ad8ac0d6-e08e-4bb1-b74c-20d7178613d3"

[INFO]  Build finished with exit code 10
```

I was able to reproduce the problem with the following command:
```
for i in {1..100}; do DEV=$(losetup --offset 1048576 --sizelimit 104857600 --find --show "/var/cache/elbe/96ec898d-5fcb-4309-b70e-bd580e3728c2/sda.img") && echo $DEV && losetup -d $DEV; done
/dev/loop0
/dev/loop2
/dev/loop0
/dev/loop2
/dev/loop0
/dev/loop2
/dev/loop0
/dev/loop0
/dev/loop2
/dev/loop0
/dev/loop0
/dev/loop2
/dev/loop0
/dev/loop2
/dev/loop0
/dev/loop0
/dev/loop2
/dev/loop0
/dev/loop0
/dev/loop0
/dev/loop0
/dev/loop2
/dev/loop0
/dev/loop2
/dev/loop0
/dev/loop0
/dev/loop2
/dev/loop0
/dev/loop0
/dev/loop2
losetup: /var/cache/elbe/96ec898d-5fcb-4309-b70e-bd580e3728c2/sda.img: failed to set up loop device: Resource temporarily unavailable
/dev/loop0
/dev/loop2
/dev/loop0
/dev/loop2
...
```

When the error occurs, I observe the following line in the kernel log:
```
[  361.804012] loop_set_status: loop0 () has still dirty pages (nrpages=1)
```

This seems to be related to a kernel bug fixed by [this](https://lore.kernel.org/patchwork/patch/1075391/) patch. I'd propose to poll for the losetup call to succeed.

bgermann, 2020-04-23:
The patch is not mainline yet. As we use the Debian kernel package patching the kernel for Elbe would mean a lot of overhead. The last Debian kernel version that does not have the corrected change would be https://snapshot.debian.org/package/linux/4.19.16-1/

Does downgrading to this version help?

closed 2020-04-30 by torbenh3

bgermann, 2020-04-30:
We have modified the call a bit with v12.4. Please check if it works for you.

martin-petzold, 2020-05-01:
Thanks. I can confirm that my builds now seem to work with 12.4.

oldiob, 2020-08-04:
With Elbe 12.5, I got the same error with amd64-grub-tighten

~~~
[CMD] losetup --offset 1048576 --sizelimit 6442450944 --find --show "/var/cache/elbe/ad624403-7801-4ed3-b08a-af7a1dc2459f/sda.img"
[CMD] mkfs.ext4  -L rfs /dev/loop0
[CMD] mount /dev/loop0 /var/cache/elbe/ad624403-7801-4ed3-b08a-af7a1dc2459f/imagemnt
mke2fs 1.44.5 (15-Dec-2018)
Discarding device blocks: done
Creating filesystem with 1572864 4k blocks and 393216 inodes
Filesystem UUID: 472b1555-e713-4608-967c-284e6c69ac88
Superblock backups stored on blocks:
        32768, 98304, 163840, 229376, 294912, 819200, 884736

Allocating group tables: done
Writing inode tables: done
Creating journal (16384 blocks): done
Writing superblocks and filesystem accounting information: done

[CMD] cp -a "/var/cache/elbe/ad624403-7801-4ed3-b08a-af7a1dc2459f/filesystems/0/." "/var/cache/elbe/ad624403-7801-4ed3-b08a-af7a1dc2459f/imagemnt/"
[CMD] umount /dev/loop0
[CMD] losetup -d /dev/loop0
[CMD] losetup --offset 6444548096 --sizelimit 1944059904 --find --show "/var/cache/elbe/ad624403-7801-4ed3-b08a-af7a1dc2459f/sda.img"
[CMD] sync
losetup: /var/cache/elbe/ad624403-7801-4ed3-b08a-af7a1dc2459f/sda.img: failed to set up loop device: Resource temporarily unavailable
[CMD] losetup --offset 6444548096 --sizelimit 1944059904 --find --show "/var/cache/elbe/ad624403-7801-4ed3-b08a-af7a1dc2459f/sda.img"
[CMD] mv "/var/cache/elbe/ad624403-7801-4ed3-b08a-af7a1dc2459f/filesystems/0"/* "/var/cache/elbe/ad624403-7801-4ed3-b08a-af7a1dc2459f/target/"
losetup: /var/cache/elbe/ad624403-7801-4ed3-b08a-af7a1dc2459f/sda.img: failed to set up loop device: Resource temporarily unavailable
[ERROR] Build failed
Traceback (most recent call last):
  File "/var/cache/elbe/devel/elbepack/asyncworker.py", line 194, in execute
    skip_pbuild=self.skip_pbuilder)
  File "/var/cache/elbe/devel/elbepack/elbeproject.py", line 686, in build
    self.targetfs.part_target(self.builddir, grub_version, grub_fw_type)
  File "/var/cache/elbe/devel/elbepack/efilesystem.py", line 417, in part_target
    grub_fw_type)
  File "/var/cache/elbe/devel/elbepack/hdimg.py", line 563, in do_hdimg
    grub_fw_type)
  File "/var/cache/elbe/devel/elbepack/hdimg.py", line 459, in do_image_hd
    fslabel, target, grub)
  File "/var/cache/elbe/devel/elbepack/hdimg.py", line 401, in create_logical_partitions
    create_label(disk, logical, lpart, fslabel, target, grub)
  File "/var/cache/elbe/devel/elbepack/hdimg.py", line 357, in create_label
    loopdev = entry.losetup()
  File "/var/cache/elbe/devel/elbepack/fstab.py", line 145, in losetup
    loopdev = get_command_out(cmd)
  File "/var/cache/elbe/devel/elbepack/shellhelper.py", line 313, in get_command_out
    raise CommandError(cmd, p.returncode)
CommandError: Error: 1 returned from Command losetup --offset 6444548096 --sizelimit 1944059904 --find --show "/var/cache/elbe/ad624403-7801-4ed3-b08a-af7a1dc2459f/sda.img"
~~~

commit dc9b6e81 (2020-04-30): debian changelog: bump version to 12.4

  * change loglevel for archive validation
  * more python3 fixes
  * fix elbe debianize packaging
  * finalize porting all examples to buster
  * fix debianize barebox
  * separate licence generation for all rootfs generated.
  * fix problems with losetup (Fixes #254)

Reviewed-by: Bastian Germann <bage@linutronix.de>