Linutronix/elbe #256: elbe v12.2  initvm submit build fails on Debian Buster host

coveritytest, 2020-02-05:
98.60% done, estimate finish Wed Feb  5 09:18:01 2020
 99.62% done, estimate finish Wed Feb  5 09:18:01 2020
Total translation table size: 0
Total rockridge attributes bytes: 210516
Total directory bytes: 962560
Path table size(bytes): 6498
Max brk space used 21e000
486832 extents written (950 MB)
[INFO] Build finished with Error
Project build was not successful, current status: build_failed
elbe control wait_busy Failed

The project will not be deleted from the initvm.
The files, that have been built, can be downloaded using:
/usr/bin/elbe control get_files --output "None" "/var/cache/elbe/db5df9e2-6ed7-4364-8c23-b65be2fcc44e"

The project can then be removed using:
/usr/bin/elbe control del_project "/var/cache/elbe/db5df9e2-6ed7-4364-8c23-b65be2fcc44e"

bgermann, 2020-02-05:
This is not useful without any information on the built project or a complete log (download the files with get_files as advised and post log.txt).

coveritytest, 2020-02-05:
I uploaded the log.txt here: https://pastebin.pl/view/f8439ae8

bgermann, 2020-02-05:
There are issues in your finetuning section: /usr/bin/sed: can't read /usr/share/initramfs-tools/scripts/nfs: No such file or directory

coveritytest, 2020-02-05:
It was indeed a finetuning problem, thanks for the hint.

closed 2020-02-05 by coveritytest