Linutronix/elbe #260: Failed to create initvm from the bin-cdrom.iso

codelectron, 2020-03-07:
Im trying to create initvm from the bin-cdrom.iso generated by an elbe build.
I started the example docker project as shown below.
`elbe initvm submit examples/x86_64-docker.xml`
which generates a lot of files including the **bin-cdrom.iso**. Now using this iso file I tried to create an initvm image as shown below. 
`elbe initvm create elbe-build-20200222-204125/bin-cdrom.iso`

and it throws the following error. 

```
[            (1*installer)  2 shell  3 shell  4- log           ][ Mar 07 21:13 ]
Configuring the network with DHCP  ... 100%                                     
Detecting disks and all other hardware  ... 95%... 100%                         
Loading additional components  ... 12%... 25%... 37%... 50%... 62%... 75%... 87%... 100%
Loading additional components  ... 25%... 50%... 75%... 100%                    
Starting up the partitioner  ... 13%... 22%... 31%... 40%... 50%... 63%... 72%... 81%... 90%... 100%
Guided partitioning  ... 20%... 40%... 60%... 80%                               
Partitions formatting  ... 33%                                                  
Partitions formatting                                                           

Installing the base system  ... 17%... 20%... 30%... 40%... 50%... 60%... 70%... 83%
    ... 91%... 100%                                                             
Configuring apt  ... 11%
Configure the package manager
-----------------------------

!! ERROR: apt configuration problem

An attempt to configure apt to install additional packages from the CD failed.
[Press enter to continue] 

... 22%... 33%... 44%... 50%... 61%... 72%... 83%... 94%... 100%
Select and install software  ... 10%... 95%
Select and install software
---------------------------

!! ERROR: Installation step failed

An installation step failed. You can try to run the failing item again from the
menu, or skip it and choose something else. The failing step is: Select and 
install software
[Press enter to continue] 
```

Additional information: I am using the Debian buster host VM and elbe 12.2.
Not sure if this issue is the same as https://github.com/Linutronix/elbe/issues/32

torbenh3, 2020-03-11:
Please go to the log screen using ctrl-a ctrl-a 4
and paste the Error from there. 

It should also be possible to scroll back using ctrl-a ctrl-a PgUp.

codelectron, 2020-03-11:
Here it is.
```

[    0.000000] Linux version 4.19.0-8-amd64 (debian-kernel@lists.debian.org) (gcc version 8.3.0 (Debian 8.3.0-6)) #1 SMP Debian 4.19.98-1 (2020-01-26)
[    0.000000] Command line: root=/dev/vda1 debconf_priority=critical console=ttyS0,115200n1 DEBIAN_FRONTEND=text
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003ffdbfff] usable
[    0.000000] BIOS-e820: [mem 0x000000003ffdc000-0x000000003fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS 1.12.0-1 04/01/2014
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 313e6001, primary cpu clock
[    0.000000] kvm-clock: using sched offset of 3133264658 cycles
[    0.000061] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000218] tsc: Detected 2807.994 MHz processor
[    0.004161] last_pfn = 0x3ffdc max_arch_pfn = 0x400000000
[    0.004455] x86/PAT: PAT not supported by CPU.
[    0.004491] x86/PAT: Configuration [0-7]: WB  WT  UC- UC  WB  WT  UC- UC  
[    0.013954] found SMP MP-table at [mem 0x000f5a40-0x000f5a4f]
[    0.015376] RAMDISK: [mem 0x3d5b6000-0x3ffcffff]
[    0.015838] ACPI: Early table checksum verification disabled
[    0.016150] ACPI: RSDP 0x00000000000F57A0 000014 (v00 BOCHS )
[    0.016191] ACPI: RSDT 0x000000003FFE1685 000030 (v01 BOCHS  BXPCRSDT 00000001 BXPC 00000001)
[    0.016335] ACPI: FACP 0x000000003FFE1549 000074 (v01 BOCHS  BXPCFACP 00000001 BXPC 00000001)
[    0.016359] ACPI: DSDT 0x000000003FFE0040 001509 (v01 BOCHS  BXPCDSDT 00000001 BXPC 00000001)
[    0.016374] ACPI: FACS 0x000000003FFE0000 000040
[    0.016378] ACPI: APIC 0x000000003FFE15BD 000090 (v01 BOCHS  BXPCAPIC 00000001 BXPC 00000001)
[    0.016382] ACPI: HPET 0x000000003FFE164D 000038 (v01 BOCHS  BXPCHPET 00000001 BXPC 00000001)
[    0.018030] No NUMA configuration found
[    0.018036] Faking a node at [mem 0x0000000000000000-0x000000003ffdbfff]
[    0.018145] NODE_DATA(0) allocated [mem 0x3ffd7000-0x3ffdbfff]
[    0.018642] Zone ranges:
[    0.018645]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.018649]   DMA32    [mem 0x0000000001000000-0x000000003ffdbfff]
[    0.018651]   Normal   empty
[    0.018653]   Device   empty
[    0.018655] Movable zone start for each node
[    0.018656] Early memory node ranges
[    0.018658]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[    0.018660]   node   0: [mem 0x0000000000100000-0x000000003ffdbfff]
[    0.035840] Reserved but unavailable: 98 pages
[    0.035862] Initmem setup node 0 [mem 0x0000000000001000-0x000000003ffdbfff]
[    0.187125] ACPI: PM-Timer IO Port: 0x608
[    0.187228] ACPI: LAPIC_NMI (acpi_id[0xff] dfl dfl lint[0x1])
[    0.187600] IOAPIC[0]: apic_id 0, version 17, address 0xfec00000, GSI 0-23
[    0.187606] ACPI: INT_SRC_OVR (bus 0 bus_irq 0 global_irq 2 dfl dfl)
[    0.187608] ACPI: INT_SRC_OVR (bus 0 bus_irq 5 global_irq 5 high level)
[    0.187610] ACPI: INT_SRC_OVR (bus 0 bus_irq 9 global_irq 9 high level)
[    0.187646] ACPI: INT_SRC_OVR (bus 0 bus_irq 10 global_irq 10 high level)
[    0.187647] ACPI: INT_SRC_OVR (bus 0 bus_irq 11 global_irq 11 high level)
[    0.187655] Using ACPI (MADT) for SMP configuration information
[    0.187657] ACPI: HPET id: 0x8086a201 base: 0xfed00000
[    0.187738] smpboot: Allowing 4 CPUs, 0 hotplug CPUs
[    0.187908] PM: Registered nosave memory: [mem 0x00000000-0x00000fff]
[    0.187910] PM: Registered nosave memory: [mem 0x0009f000-0x0009ffff]
[    0.187912] PM: Registered nosave memory: [mem 0x000a0000-0x000effff]
[    0.187913] PM: Registered nosave memory: [mem 0x000f0000-0x000fffff]
[    0.187918] [mem 0x40000000-0xfeffbfff] available for PCI devices
[    0.187920] Booting paravirtualized kernel on KVM
[    0.187957] clocksource: refined-jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645519600211568 ns
[    0.324697] random: get_random_bytes called from start_kernel+0x93/0x52c with crng_init=0
[    0.324707] setup_percpu: NR_CPUS:512 nr_cpumask_bits:512 nr_cpu_ids:4 nr_node_ids:1
[... 204 lines cut ...]
[    4.695145] mip6: Mobile IPv6
[    4.697489] NET: Registered protocol family 17
[    4.702067] mpls_gso: MPLS GSO support
[    4.707665] mce: Using 10 MCE banks
[    4.710644] sched_clock: Marking stable (4154817776, 555540722)->(5440278331, -729919833)
[    4.731547] registered taskstats version 1
[           1- installer  2 shell  3 shell   (4*log)           ][ Mar 11 17:49 ]
Mar 11 17:47:31 apt-setup: Copying package lists...                             
Mar 11 17:47:31 apt-setup: gpgv: Signature made Sat 22 Feb 2020 10:28:32 AM UTC 
Mar 11 17:47:31 apt-setup: gpgv:                using RSA key 7E521ACEA77065575789B2C2AC2E59C495498AD2                                                          
[... cut ...]
Mar 11 17:47:31 apt-setup: Repeat this process for the rest of the CDs in your set.                                                                             
Mar 11 17:47:31 apt-setup: E: Sub-process gpgv returned an error code (2)       
Mar 11 17:47:31 apt-setup: W: Signature verification failed for: /media/cdrom/dists/buster/InRelease                                                            
Mar 11 17:47:31 kernel: [  242.121460] ISO 9660 Extensions: Microsoft Joliet Level 3                                                                            
Mar 11 17:47:31 kernel: [  242.123410] ISO 9660 Extensions: RRIP_1991A          
```

oldiob, 2020-05-06:
The elbe-keyring.gpg file is of type "GPG keybox database version 1".  It should be of type "PGP/GPG key public ring (v4)".

oldiob, 2020-07-21:
This should be fix on master. See 7bec1291a75e4db2226fa12527070916a718772c

commit 67e52b03 (2020-07-22): debian: bump elbe version to v12.5

  * Fix building sdk for amd64
  * Add cross build support to pbuilder.
  * Add ccache support to pbuilder.
  * Limited finetuning support for initvm.
  * First part of the elbe test suite
    * doctests for filesystem
    * doctests for shellhelper
    * pylint as unittest
  * schema: Add possibility to add multiple binary blobs
  * debianize: kernel: Add lzop build dependency
  * initvmaction: Add sync action
  * Include Source Packages for Built-Using: Entries into src-cdrom.
  * src-cdrom now sorts packages in components for sysroot, host-sysroot,
    ta [... cut]
closed 2020-07-22 by torbenh3