Linutronix/elbe #261: elbe initvm create starts virtual console

joao404, 2020-03-11:
Hello,

I am using elbe in an automated building environment. The current used elbe version is 2.1 which I want to update to 12.2. The problem is that by executing "elbe initvm create" a virtual terminal is started which loggs everything in a 24x80 console inside my own shell. This makes logging for me not possible. Can this second virtual terminal be deactivated? 
I searched for a while in the settings of qemu but found nothing that works.

[     (1*installer) 2 shell  3 shell  4- log           ][ Mar 11 15:09 ]
.....

Thanks
Marcel

matlec, 2020-04-23:
We use "script" to wrap the output in our builds:
script --return --quiet -c "<your elbe command>" /dev/null

oldiob, 2020-06-09:
The initial image is run with 'qemu-system -nographic'.  There's is no second virtual terminal.  QEMU simply redirect its console to your terminal.  The virtual terminal you're talking about is screen which I guess is used by the Debian installer.

Wrapping the call to elbe like @matlec proposed is a good solution.  Optionally, you can use the '-serial dev' option to redirect the communication of QEMU to somewhere else.

closed 2021-03-17 by bgermann