Linutronix/elbe #263: For nested virtualization user needs to be in group 'kvm'

martin-petzold, 2020-05-01:
As far as I remember the documentation mentioned that for nested virtualization the user needs to be in group 'kvm' (next to 'libvirt' and 'libvirt-qemu'). However, it seems this was removed from current documentation. I would suggest to add this again.

bgermann, 2020-05-01:
It is still there (docs/quickstart.txt)

closed 2020-05-01 by bgermann

martin-petzold, 2020-05-01:
Sorry, it is not there. You only talk about:

1. "This allows using hardware accelerated virtualization implemented by kvm."
2. "When Debian is running inside a VM (vmware etc), you need to make sure, that nested KVM is working."

There is no explicit hint, that the user needs to be in the group "kvm" for nested virtualization to work.

bgermann, 2020-05-01:
https://github.com/Linutronix/elbe/blob/master/docs/quickstart.txt#L87

martin-petzold, 2020-05-01:
Okay, but "libvirt" =! "kvm". The user needs to be in "libvirt" AND "kvm". On Debian Buster.

oldiob, 2020-05-01:
Agree.  I remember I had to look around why KVM was not enabled on my initvm.  This is a one-liner and could save time for new users.