Linutronix/elbe #264: elbe fails with a fresh install on Debian 10 (buster) and the elbe from git

Collin01, 2020-05-04:
Elbe fails on a fresh install on debian10.

Steps to reproduce:
- Install debian 10 in a vm.
- install packages: 
  apt install python python-debian python-mako python-lxml python-apt python-gpg python-suds python-libvirt qemu-utils qemu-kvm p7zip-full make
- git clone https://github.com/Linutronix/elbe.git
- cd elbe
- ./elbe initvm create ./examples/armhf-ti-beaglebone-black.xml

it produce this output:
(Please see attachment)
[log.txt](https://github.com/Linutronix/elbe/files/4574350/log.txt)

codelectron, 2020-05-04:
I think you are missing the libvirt related installation and further configuration steps like here

https://elbe-rfs.org/docs/sphinx/article-quickstart.html?highlight=adduser#create-initvm-and-submit-xml-files

oldiob, 2020-05-05:
See this issue <https://github.com/Linutronix/elbe/issues/248>

bgermann, 2021-03-17:
There was no feedback and therefore I am closing this as duplicate.

closed 2021-03-17 by bgermann