Linutronix/elbe #268: elbe initvm create fail after a fresh Debian 10 install - "Checking the Debian archive mirror"

anthonyemtrion, 2020-05-12:
Hello, 

I'm trying to install an initvm on my laptop from a fresh Debian 10 install and elbe failed at the step "Checking the Debian archive mirror". 
I can ping deb.debian.org before lauching the command "elbe initvm create" and there is no problem. 
I've followed step-by-step the "Quick start" documentation. 
I've also git clone this repo and use the master branch to test and it doesn't work either. 
I've got only my Ethernet and no Wifi activated on my laptop. 
I've made a record from my Terminal to check everything slowly until it prompt "!!!ERROR bad mirror...". Changing mirror has suggested by the installation at the end and does not help either.  
I've checked in Wireshark, my ethernet connection and filtering by the MAC address I found in the ./initvm/libvirt.xml and I see nothing. I've checked either the TCP port 7587 and nothing. 

I've got no idea where else to look. This a fresh install, I imagined it should work out of the box! 

$ cat /etc/debian_version 
10.4
elbe v12.4

Best regards, 
Anthony

codelectron, 2020-05-12:
Hi,

The source of the initvm recipe is located here https://github.com/Linutronix/elbe/blob/master/elbepack/init/default-init.xml
Make a copy of it and change the mirror to some other and try again. 
You can create initvm with custom XML like this
`elbe initvm create custom-initvm-recipe.xml`

The deb.debian.org  redirects to the fastest/nearest mirror,provide a actual mirror site from the list and see if it works https://www.debian.org/mirror/list

oldiob, 2020-05-12:
Hello,

In the initvm's installer, you have 4 sessions (taking tmux definition of session here).  By default you're in session 1.  You can switch between session using "C-a C-a [1-4]".

The fourth session has a more verbose log of the installer.  The second and third session are shell (ash).  

Could you switch to one of these shell and try to reach the mirror?  I'm not sure if ping is available at this point.  Anyway, try to see if you can send packet on the network and get a response.

Also, from the same shell you can 'more' the log IIRC.  Don't remember which file tho.

anthonyemtrion, 2020-05-12:
Hello @codelectron @oldiob , 

Thank both of you for your help!

So I managed to change the mirror by using a custom XML. I doubt this would be the problem. The default Mirror is the FTP of Germany and I'm living not really far from it. Anyway I changed to the FTP in France. 

So I checked the log and what I found : 
May 12 12:05:54 main-menu[2621]: INFO: Menu item 'choose-mirror' selected       
May 12 12:05:54 anna-install: Queueing udeb apt-mirror-setup for later installation                                                                             
May 12 12:05:54 choose-mirror[3523]: DEBUG: command: wget --no-verbose http://ftp.fr.debian.org/debian/dists/oldstable/Release -O - | grep -E '^(Suite|Codename|Architectures):'              
May 12 12:06:14 choose-mirror[3523]: DEBUG: command: wget --no-verbose http://ftp.fr.debian.org/debian/dists/stable/Release -O - | grep -E '^(Suite|Codename|Architectures):'                 
May 12 12:06:34 choose-mirror[3523]: DEBUG: command: wget --no-verbose http://ftp.fr.debian.org/debian/dists/testing/Release -O - | grep -E '^(Suite|Codename|Architectures):'                
May 12 12:06:55 choose-mirror[3523]: DEBUG: command: wget --no-verbose http://ftp.fr.debian.org/debian/dists/unstable/Release -O - | grep -E '^(Suite|Codename|Architectures):'               
May 12 12:07:15 choose-mirror[3523]: INFO: mirror does not have any suite symlinks                                                                              
May 12 12:07:15 choose-mirror[3523]: DEBUG: command: wget --no-verbose http://ftp.fr.debian.org/debian/dists/buster/Release -O - | grep -E '^(Suite|Codename|Architectures):'                 
May 12 12:07:35 choose-mirror[3523]: WARNING **: mirror does not support the specified release (buster) 

Now I see the real problem but this is quite strange...

anthonyemtrion, 2020-05-12:
other than that: 
~ # ip addr                                                                     
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1000                 
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00                       
    inet 127.0.0.1/8 scope host lo                                              
       valid_lft forever preferred_lft forever                                  
    inet6 ::1/128 scope host                                                    
       valid_lft forever preferred_lft forever                                  
2: ens4: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000  
    link/ether 1a:89:47:c8:98:5c brd ff:ff:ff:ff:ff:ff                          
    inet 10.0.2.15/24 scope global ens4                                         
       valid_lft forever preferred_lft forever                                  
    inet6 fec0::1889:47ff:fec8:985c/64 scope site dynamic                       
       valid_lft 86185sec preferred_lft 14185sec                                
    inet6 fe80::1889:47ff:fec8:985c/64 scope link                               
       valid_lft forever preferred_lft forever                                  
~ #  
~ #                                                                             
~ #                                                                             
~ # ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
^C
--- 8.8.8.8 ping statistics ---
72 packets transmitted, 0 packets received, 100% packet loss
~ # 
~ # 
~ # udhcpc ens4
udhcpc: SIOCGIFINDEX: No such device
~ # 

That doesn't lok good at all....

oldiob, 2020-05-12:
Does 
~~~sh
$ wget --no-verbose http://ftp.fr.debian.org/debian/dists/buster/Release -O - | grep -E '^(Suite|Codename|Architectures):'
~~~
results in 
~~~sh
Suite: stable
Codename: buster
Architectures: amd64 arm64 armel armhf i386 mips mips64el mipsel ppc64el s390x
~~~
in your shell?

anthonyemtrion, 2020-05-12:
On my Host computer with Debian 10, yes it does. 
Inside the VM: 
~ # 
~ # wget --no-verbose http://ftp.fr.debian.org/debian/dists/buster/Release -O - 
| grep -E '^(Suite|Codename|Architectures):'
wget: unable to resolve host address 'ftp.fr.debian.org'
~ # 

Something is wrong with the network for sure, DNS doesn't work.

anthonyemtrion, 2020-05-12:
Is it not related to this : https://wiki.debian.org/QEMU#Networking ? 

Do we need a bridge to make it work ? because my Guest has the IP address 10.0.2.15 and this is internal DHCP from QEMU. 
But how does it know how to resolve DNS ? he needs a bridge to the Host for that no ?
Sadly the Busybox inside the initvm is really bare minimal and I can't run iproute.

codelectron, 2020-05-12:
Some information for clarity.
- The InitVM gets internet out of the box theres nothing that needs to be done there (from my experience).  
- The busybox is actually part of the installer and its not InitVM. When you have successcully installed InitVM then you get a full fledged Debian OS.

anthonyemtrion, 2020-05-14:
Thanks for your help @codelectron and @oldiob but I found the problem. 
I'm not even 100% sure this was the problem but on my internet box DHCP server I selected the "Force Broadcast response" and change the DNS server address. I gave only one address. 
I was able after that to create the initvm. The problem was on my network configuration.

closed 2020-05-14 by anthonyemtrion