Linutronix/elbe #273: How to cache downloaded files?

Ecordonnier, 2020-07-14:
Did someone manage to integrate a tool like apt-cacher-ng/squid to the build process in order to speed-up elbe build time? At the moment, we are repeatingly downloading the same debian package files for each build, making the build process very inefficient.

bgermann, 2020-07-14:
Just use your apt-cacher-ng repository as primary mirror. There is no need for an integration in Elbe.

Ecordonnier, 2020-07-14:
Well, elbe provides the Dockerfile to run the build which could include apt-cacher-ng if it is useful for many users, and there is an open issue against apt-cacher-ng which makes me think it may not work (https://github.com/Linutronix/elbe/issues/121). I'll try to configure it as main mirror and configure apt-cacher-ng's caching path mounted as a docker volume, but it would have been useful to know if other people already tried and have a working setup to cache download.

Ecordonnier, 2020-07-14:
I tried to make it work, but it is not that trivial with an HTTPS mirror. Firstly I replaced the mirror used with the syntax required by apt-cacher-ng which is "http://localhost:3142/HTTPS///mydebianmirror/debian-package-mirror". Elbe manages to download all the packages but this is problematic because when configuring "http" as "primary_proto", then apt-transport-https does not get installed by debootstrap any more and elbe fails with "Error Updating rpcaptcache: E:The method driver /usr/lib/apt/methods/https could not be found.".

Debootstrap has an option "--include" which would allow to add "apt-transport-https" manually, but the issue is that elbe allows to modify this option only when specifying a "debootstrapvarianttype" tag, and I didn't yet find how to add the package  "apt-transport-https" without changing the debootstrap variant.

Ecordonnier, 2020-07-15:
I managed by doing the following, it is probably interesting for other people so I'll post it there:

In Dockerfile, install package apt-cacher-ng, and use those lines:
```
COPY acng.conf /etc/apt-cacher-ng/
CMD mkdir -p /apt-cacher-ng-cache && chmod 777 /apt-cacher-ng-cache && /etc/init.d/apt-cacher-ng start && /bin/bash
```

In acnf.conf, configure your corporate proxy correctly if you have one (Proxy: http://my-proxy) as wel as the CacheDir option. Make sure that the CacheDir is mounted as a docker volume while running docker, otherwise it will be lost after each build.

If you are using an https debian mirror, you need to use the following syntax in the configuration:

```
    <primary_host>localhost:3142</primary_host>
    <primary_path>HTTPS///my-debian-mirror</primary_path>
    <primary_proto>http</primary_proto>

```

Also, you need to make sure that all your debian mirrors are using http as a protocol, if you have extra URLs defined in "url-list" with https mirrors, you will get the error I was getting (package apt-transport-https missing).

It reduces my elbe build-times from 18 minutes to 11 minutes.

closed 2020-07-15 by Ecordonnier