Linutronix/elbe #274: Elbe wrongly replaces /etc/resolv.conf

hdengler, 2020-07-15:
### Issue
The Debian package ``resolvconf`` creates a symbolic link for ``/etc/resolv.conf``. In an Elbe build (tested with Elbe 12.4), this link will be silently replaced by a file (containing the nameserver configuration for qemu). As a workaround, the link can be re-created in finetuning section.

I attached an example xml (``project.xml.txt``) to this issue, so that this issue can be easily reproduced. 

Building the xml without specifying a variant, you get an issue.tar.gz with an wrongly replaced /etc/resolv.conf, although the log.txt states:
```
[...]
[INFO] |+/etc/resolv.conf+|postinst generated
[...]
```

### Workaround
The attached xml has also a workaround for that problem. Building the xml with ``--variant=workaround`` will generate a RFS with a fixed link for ``/etc/resolv.conf``.
```
[...]
[INFO] |+/etc/resolv.conf+|modified finetuning
[...]
```

[project.xml.txt](https://github.com/Linutronix/elbe/files/4925197/project.xml.txt)

oldiob, 2020-07-15:
This is the correct way to protect /etc/resolv.conf.  Some finetuning commands will protect the modified files from the excursion in the RFS.  ln is one of these.

hdengler, 2020-07-20:
No, the Elbe behavior is wrong. 

1. the postinst of the Package `resolvconf` creates a symbolic link for `/etc/resolv.conf` (see the first log entry).
2. Elbe replaces this generated `/etc/resolv.conf` with its own version **without any log entry or an update it in the file list!**

The replaced file contains the internal qemu nameserver configuration, but the file list in the log.txt contains states, that this file is generated by the postint of resolvconf. That is **not** correct!

In my opinion it is a bug, if Elbe **silently** replace a file with its own variant, and I have to fix it in the finetuning section.

bgermann, 2022-08-15:
A side note on this: There is some recent change in Debian sid that deprecates the resolv.conf file in some way. I noticed this on updating a Debian sid installation yesterday. Please investigate this and figure out if we need any changes in Elbe.

hdengler, 2022-08-15:
I'll have a look at this the next time, I'm setting up a new system.

Anyhow, the core problem of this issue is, that elbe replaces the file `/etc/resolv.conf`, without checking if it exists or not. The bug is in my opinion in commit 5c2ae6921ba858273a81f989c8da3973ad0f99f4, which implements the Excursion class. In function _undo_excursion(self, rfs), the original file is replaced without any warning or error. In my opinion, this is a bug.

t-8ch, 2024-04-29:
There is a tentative fix for this issue in https://lists.linutronix.de/pipermail/elbe-devel/2024-April/006859.html

Testers would be welcome.

closed 2024-05-07 by t-8ch