Linutronix/elbe #278: XML Validation Error in armhf-ubuntu.xml example

smartdabao, 2020-08-27:
Error happens during initvm create examples/armhf-ubuntu.xml, error info is as follows.

elbe control set_xml failed2
Server returned error:

Elbe XML Validation Error
/var/cache/elbe/6b4044f8-cb2e-45e3-b15b-4dc2bbd35a22/source.xml:21 error Element 'options': This element is not expected.
/var/cache/elbe/6b4044f8-cb2e-45e3-b15b-4dc2bbd35a22/source.xml:28 error Element 'options': This element is not expected.
/var/cache/elbe/6b4044f8-cb2e-45e3-b15b-4dc2bbd35a22/source.xml:35 error Element 'options': This element is not expected.
/var/cache/elbe/6b4044f8-cb2e-45e3-b15b-4dc2bbd35a22/source.xml:42 error Element 'options': This element is not expected.

Giving up

oldiob, 2020-08-27:
What version of Elbe are you using?  Are you using the master branch?  It seems like your host version doesn't match the initvm version of Elbe.

The <options> tag was added to Elbe recently.  To stay compatible with old version, Elbe removes embedded mirror's options (e.g. [arch=armhf]), and replaces them with the <option> tags in your XML.

In my opinion, your initvm is not using the current Elbe XML schema and of course this result in a parsing error of your source.xml on the **server side**.  Thus, try to make your initvm match the version of your host installation.

smartdabao, 2020-08-27:
Thank you, I remove [arch=armhf] options in xml file, it works. I think the example armhf-ubuntu.xml file in master branch need to be updated.

closed 2020-08-27 by smartdabao

oldiob, 2020-08-27:
No the example is fine.  Like I said, Elbe is still compatible with the old way, **you should not remove the mirror's options**.

IIRC, if you remove the mirror's options for the Ubuntu example, you will get some problem with Apt later on.