Linutronix/elbe #279: How to change repository

smartdabao, 2020-09-01:
It is not a bug, but I do not know where to ask questions, so I post it here. 
I want to use 'ftp.cn.debian.org' instead of 'ftp.de.debian.org' to speed up the process. But when I modify primary_host tag in xml, the initvm create command still download initrd-cdrom.gz from 'ftp.de.debian.org'.
How to change it? Do I have to replace 'ftp.de.debian.org' with 'ftp.cn.debian.org' in the whole elbe project?
Thank you.

bgermann, 2020-09-01:
Look at example elbe-init-big-machine.xml which has an initvm XML section which holds the mirrors for the initvm. After that section you can have project and target sections as with the other examples.

smartdabao, 2020-09-01:
Thank you for your reply.

closed 2020-09-01 by smartdabao