Linutronix/elbe #281: pbuilder: debhelper-compat support broken in 12.5

jogness, 2020-09-03:
The Debian compat version can be specified as a build dependency in debian/control:

`Build-Depends: debhelper-compat (= 12)`

With elbe 12.5 the building of such packages using elbe pbuilder breaks with an error about not finding the version. debhelper-compat is a virtual package and this syntax is used to specify the compat level.

An example of such a package is mate-session-manager.

This works with elbe 12.4.

jogness, 2020-10-01:
With elbe 13.0, "elbe pbuilder" still complains about the compatibility specification:

    E: Version '12' for 'debhelper-compat' was not found
       -> =========================================
       -> (not adding  to debhelper-compat=12)
           -> Cannot install debhelper-compat=12; apt errors follow:
    Reading package lists...
    Building dependency tree...
    Reading state information...
    E: Version '12' for 'debhelper-compat' was not found
         -> Considering debhelper to satisfy the dependency

However, now it is downloading the built packages at the end instead of aborting with an error. I don't know if elbe is actually respecting the compatibility specification, but at least elbe 13.0 will build the Debian packages again.

I didn't find and commit that addressed this issue directly. Did this problem disappear on its own or did someone actually look into this?

bgermann, 2020-10-01:
I guess this was a side effect of fixing "pbuilder: fix adding built packages to internal repo"

closed 2020-10-08 by bgermann