Linutronix/elbe #282: debianize: kernel: arch missmatch while stripping

yegorich, 2020-09-04:
I am trying to cross-compile the kernel using `dpkg-buildpackage` and get the following error:

```
   dh_strip_nondeterminism
   dh_compress
   dh_fixperms
   dh_missing
   dh_strip
arm-linux-gnueabihf-objcopy:debian/linux-headers-5.4.61-elbe/usr/src/linux-headers-5.4.61-elbe/scripts/unifdef: File format not recognized
dh_strip: arm-linux-gnueabihf-objcopy --only-keep-debug --compress-debug-sections debian/linux-headers-5.4.61-elbe/usr/src/linux-headers-5.4.61-elbe/scripts/unifdef debian/.debhelper/linux-headers-5.4.61-elbe/dbgsym-root/usr/lib/debug/.build-id/66/aea5f31c4da718e283c406a270627928e5eeec.debug returned exit code 1
dh_strip: Aborting due to earlier error
debian/rules:89: recipe for target 'binary' failed
make: *** [binary] Error 2
dpkg-buildpackage: error: fakeroot debian/rules binary subprocess returned exit status 2
Command exited with non-zero status 2
```

Looks like it tries to strip the native binaries.

yegorich, 2020-09-04:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=753507

bgermann, 2020-09-04:
Okay. Want to hand that in as a patch on elbe-devel?

yegorich, 2020-09-04:
It looks like this Debian bug report fixes only the `unifdef` package. The kernel has its own version in `scripts` and this is our problem.

If the strip step is skipped, the problem appears later when you try to compile an out-of-tree driver with these kernel headers as the scripts were compiled for x86 and not the target arch.

How can we fix this?

bgermann, 2022-08-24:
Just to be clear: Does this apply to a kernel debianization that you have created with Elbe? Else this is not an Elbe bug.

bene42, 2023-10-25:
elbe debianize is droped.

closed 2023-10-25 by bene42