Linutronix/elbe #283: Traceback: Building image doesn't find previously generated package

schlatterbeck, 2020-09-14:
I'm successfully (after applying patch for gitub issue #280) building a package using elbe pbuilder.
This used to insert the newly-built package into the package cache. This worked in previous versions but does no longer in 12.5+deb10+build901

The call
```
elbe control --build-bin build $(cat elbe.prj)
elbe control wait_busy $(cat elbe.prj)
```

fails with the following traceback (my package successfuly built with pbuilder for the same project is called iptna):
```
[ERROR] Build failed
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/elbepack/asyncworker.py", line 194, in execute
    skip_pbuild=self.skip_pbuilder)
  File "/usr/lib/python2.7/dist-packages/elbepack/elbeproject.py", line 617, in build
    self.get_rpcaptcache())
  File "/usr/lib/python2.7/dist-packages/elbepack/efilesystem.py", line 105, in extract_target
    deps = cache.get_dependencies(p)
  File "<string>", line 2, in get_dependencies
  File "/usr/lib/python2.7/multiprocessing/managers.py", line 774, in _callmethod
    raise convert_to_error(kind, result)
KeyError: "The cache has no package named 'iptna'"

```

JensHillenstedt, 2020-09-15:
I am having the same issue here while trying to include the debhello package created by following the Guide For Debian Maintainers. I converted the package via gbp into a git repository. At the end of `elbe pbuilder build --project ${PROJECT}` i get this output

> Warning: database 'buster|main|armhf' was modified but no index file was exported.
Changes will only be visible after the next 'export'!
Warning: database 'buster|main|source' was modified but no index file was exported.
Changes will only be visible after the next 'export'!
[CMD] reprepro --basedir /var/cache/elbe/c060ad7c-7f86-44f0-b129-42c3e83b3995/repo removesrc buster debhello
[CMD] reprepro --basedir "/var/cache/elbe/c060ad7c-7f86-44f0-b129-42c3e83b3995/repo" remove buster debhello
Exporting indices...
Deleting files no longer referenced...
Not removed as not found: debhello
Exporting indices...
[CMD] reprepro --basedir "/var/cache/elbe/c060ad7c-7f86-44f0-b129-42c3e83b3995/repo" export buster
[INFO] Pdeb finished successfully

bgermann, 2021-03-17:
Is this true for the current Elbe version 13.2 that comes with that patch and runs with Python 3?