Linutronix/elbe #284: Link to dbsfed.xsd is a hard-coded https link that leads to a 404 error

schlatterbeck, 2020-09-16:
When  for downgrading I'm installing elbe releases/v12.4 from git, the two files `elbepack/validate.py` and `elbepack/xmlpreprocess.py` contain a hard-coded link:

```
schema_file = "https://www.linutronix.de/projects/Elbe/dbsfed.xsd"
```

which currently produces a 404 error. I've verified that the same code is on the current master branch, too.
Since a working version of that file is in the github repository in `schema/dbsfed.xsd` I'd propose to replace the link with the path of the installed version of that file (it currently doesn't seem to be installed).

bgermann, 2021-03-08:
The namespace is not supposed to direct to a valid web location. On installing elbe-schema, the XML should look up that file from the locally installed XML catalogs.

closed 2021-03-08 by bgermann

bgermann, 2021-03-08:
related: #115