Linutronix/elbe #285: Additional debootstrap packages are not possible if using standard debootstrap variant

pzimmermann-dh, 2020-10-01:
Hello,
when using ELBE, I came across the following problem:

If you want to use a Debian repository, which is secured by TLS (e.g. Docker's repsotitory), at a ELBE build you have to use the "includepkgs" attribute of the debootstrapvariant tag to add "apt-transport-https" (which is not needed anymore in buster or later) and "ca-certificates".

The problem is that you have to define a non-default debootstrapvariant like minbase if you include the debootstrapvariant tag inside the XML.
Thus it isn't possible to include the needed packages for using HTTPS repos for debootstrap if you want to use the default debootstrap variant.

bgermann, 2020-10-01:
This is a known issue and we are happy to take patches for it. In the mean time just add the package diff to your minbase recipe. You can find the list via `aptitude search '?priority(important)'`

pzimmermann-dh, 2020-10-01:
Thank you for your answer, we will use adding the package diff to the recipe as a workaround.

closed 2021-03-08 by bgermann

dbraunwarth, 2022-01-04:
Fixed by eb942a624040553dcb95f111db2386dff822e87a