Linutronix/elbe #286: Fail to build in docker

smartdabao, 2020-10-08:
I add my user to docker group, I can access the elbe docker container, just follows instructions through README.
In elbe docker container, I type the command "elbe initvm create examples/armhf-ti-beaglebone-black.xml", and I got error.
 98.98% done, estimate finish Wed Oct  7 14:26:11 2020
 99.53% done, estimate finish Wed Oct  7 14:26:12 2020
Total translation table size: 0
Total rockridge attributes bytes: 292816
Total directory bytes: 1387312
Path table size(bytes): 9184
Max brk space used 2f9000
904231 extents written (1766 MB)
[CMD] mv /var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749/chroot/repo /var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749/chroot/../
[CMD] rm /var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749/chroot/etc/apt/sources.list.d/local.list
[CMD] rm /var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749/chroot/etc/apt/trusted.gpg.d/elbe-localrepo.gpg
[CMD] losetup --find --show --partscan "/var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749/sdcard.img"
[CMD] dd if=/dev/loop0p1 of="/var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749/scard1.img"
102400+0 records in
102400+0 records out
52428800 bytes (52 MB, 50 MiB) copied, 1.56857 s, 33.4 MB/s
[CMD] dd if=/dev/loop0p2 of="/var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749/scard2.img"
2967552+0 records in
2967552+0 records out
1519386624 bytes (1.5 GB, 1.4 GiB) copied, 27.9788 s, 54.3 MB/s
[CMD] losetup --detach "/dev/loop0"
[CMD] mv -v /var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749/target/usr/lib/u-boot/am335x_boneblack/MLO ../MLO
renamed '/var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749/target/usr/lib/u-boot/am335x_boneblack/MLO' -> '../MLO'
[CMD] tar cvzf "/var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749/sdcard.img.tar.gz" --sparse -C "/var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749" "sdcard.img"
sdcard.img
[CMD] rm -f "/var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749/sdcard.img"
[CMD] tar cvzf "/var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749/scard1.img.tar.gz" --sparse -C "/var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749" "scard1.img"
scard1.img
[CMD] rm -f "/var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749/scard1.img"
[CMD] tar cvzf "/var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749/scard2.img.tar.gz" --sparse -C "/var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749" "scard2.img"
scard2.img
[CMD] rm -f "/var/cache/elbe/85c33aa8-ab87-44f2-adea-477ec286c749/scard2.img"
[INFO] Build finished with Error
Project build was not successful, current status: build_failed
elbe control wait_busy Failed

It seems built was successful, and error happens when copy the files. If I copy the files manually that have been built, I can get these files.
Is it an issue or I miss something? Thank you.

bgermann, 2020-10-08:
Please upload the complete log.txt.

smartdabao, 2020-10-08:
[log.txt](https://github.com/Linutronix/elbe/files/5347915/log.txt)

bgermann, 2020-10-08:
There are some binary or source packages which cannot be downloaded:
[ERROR] Package 'libss2-1.44.5-1+deb10u3' could not be downloaded
[ERROR] Package 'libtasn1-6-4.13-3' could not be downloaded
[ERROR] Source for package 'openssh-1:7.9p1-10+deb10u2' could not be downloaded: Can't fetch item /var/cache/elbe/sources/openssh_7.9p1.orig.tar.gz.asc: Connection failed
[ERROR] Source for package 'pcre2-10.32-5' could not be downloaded: Can't fetch item /var/cache/elbe/sources/pcre2_10.32.orig.tar.gz: Connection failed

Please try a different mirror.

closed 2020-10-08 by smartdabao