Linutronix/elbe #291: SKD build fails for i386 and Debian Stretch

shifty91, 2021-01-27:
The SDK generated for a i386 platform based on Debian Stretch does not contain a cross compiler. This is most likely because the gcc package `g++-i686-linux-gnu` is not available for Debian Stretch.

Elbe: v13.1
Buildtype: i386
Suite: Stretch

bgermann, 2021-03-05:
Is this a regression or was that never tried before?

For this issue @oldiob doesn't think there's much options.  Here's what can be
done:

 A) Don't use stretch
 
 B) Add a mirror that has g++-i686-linux-gnu
 
 C) Add some very specific code to Elbe to handle that case and install
 g++ instead of g++-i686-linux-gnu.  This requires renaming all x86_64
 tools for i686 in the SDK and using the -m32 option for CC and LD.  I
 don't think this is a viable option, but it could be done with some
 script magics.

oldiob, 2021-03-05:
Note on that last solution.  

You can simply use a buildtype for amd64 instead of i686.  From there, add the option `-m32` to CFLAGS when configuring autotools.  You also need to rename everything starting with x86_64 to i686 in the toolschain.  Otherwise the dynamic linker will get confuse.

I said that this option was not viable.  By that I meant that I don't think we should handle this specific case in Elbe.

bgermann, 2021-03-10:
For C), Preventing the error and install amd64 compiler can be done by adding hostsdk-pkg-list:

```
<target>
  <hostsdk-pkg-list>
    <pkg>g++</pkg>
    <pkg>gdb-multiarch</pkg>
  </hostsdk-pkg-list>
</target>
```

bgermann, 2021-03-17:
C) and D) would need additional patches in Elbe. For option B) use hostsdk-pkg-list with gdb-multiarch and g++-6-i686-linux-gnu from [gcc-6-cross >= 30](https://snapshot.debian.org/package/gcc-6-cross/). stretch's last version v24 has gcc 6.3, v30 has gcc 6.4, v31+ have gcc 6.5. So v30 should be the best candidate.

closed 2021-03-17 by bgermann