Linutronix/elbe #292: SKD build fails for amd64 and Debian Buster

shifty91, 2021-01-27:
The cross compiler in a SDK generated for amd64 based on Debian Buster segfaults on execution.

Elbe: v13.1
Buildtype: amd64
Suite: Buster

bgermann, 2021-03-10:
From @torbenh3:

When the target arch is arm or something, we know, that ld-linux.so works fine. So we just make sure, that the treatment of ld-linux.so is identical to what is done for an arm SDK...

In elbepack/makofiles/toolchain-shar-extract.sh.mako we see that native_elf_files and target_elf_files is identical.
For the native elf files, ld-linux.so is skipped.
For the target thing it's not.

So for the non-cross case, we should probably just make target_* empty sets. Or also use the ignore code, when we fix the target_elf_files.

I am not sure, why we actually search for target files in the host sysroot. But some are probably there. 
And i would think, that when the target ld-linux.so is also in the host sysroot, then we would get 2 filenames in dl_path.

bgermann, 2021-03-10:
From @oldiob:

> use the ignore code, when we fix the target_elf_files.

This makes ld-linux work correctly.  However now `ld` is not honoring `--sysroot`.

x86_64-linux-gnu-gcc --sysroot=/tmp/sdk/sysroots/target  -O2 -pipe -g \
-feliminate-unused-debug-types -Wl,-O1 -Wl,--hash-style=gnu \
-Wl,--as-needed -Wl,--sysroot=/tmp/amd64/sysroots/target -Wl,--verbose \
hello.c
----------------------------------------------------------------------
attempt to open /lib/x86_64-linux-gnu/libc.so.6 failed
attempt to open/tmp/amd64/sysroots/host/usr/lib/x86_64-linux-gnu/libc_nonshared.a succeeded
(/tmp/amd64/sysroots/host/usr/lib/x86_64-linux-gnu/libc_nonshared.a)elf-init.oS
attempt to open /lib/x86_64-linux-gnu/ld-linux-x86-64.so.2 failed
/tmp/amd64/sysroots/host/usr/bin/ld: cannot find /lib/x86_64-linux-gnu/libc.so.6
/tmp/amd64/sysroots/host/usr/bin/ld: cannot find /lib/x86_64-linux-gnu/ld-linux-x86-64.so.2
collect2: error: ld returned 1 exit status
----------------------------------------------------------------------

bgermann, 2021-03-12:
@oldiob Can you please hand in a segfault-avoiding patch?

bgermann, 2021-03-15:
I cannot reproduce @oldiob's issue with the environment set by the generated environment file.

oldiob, 2021-03-15:
Can you run `elbe check-build sdk` on your build to ensure that the toolchain works correctly?

Can you also send me via email the XML that you've used for your build.  I'm wondering why sysroot is not honored on my system.

commit 5b2f4366 (2021-03-17): Prevent SDK segfault for amd64 targets

All executables in an SDK generated for amd64 based on Debian Buster
segfault on execution. Skip elf-patching the rpath of ld-linux.so.

Fixes #292.

Co-developed-by: Olivier Dion <dion@linutronix.de>

Reviewed-by: Torben Hohn <torben.hohn@linutronix.de>
Tested-by: Christian Teklenborg <chris@linutronix.de>
closed 2021-03-17 by torbenh3