Linutronix/elbe #296: grub legacy uses last partition to boot from

mrelyb, 2021-03-10:
grub-legacy installer in grub-installer97 loops through all fstab entries, and uses the last entry to modify menu.lst. This results in always booting from last partition, instead of booting partition marked with "grub-install" in partitions.

          for entry in self.fs.depthlist():
                do('mount /dev/mapper/poop%sp%d %s' %
                   (loopnum, entry.partnum, imagemntfs.fname(entry.mountpoint)))

---> entry is last entry of fstab

            do("mount --bind /dev %s" % imagemntfs.fname("dev"))
            do("mount --bind /proc %s" % imagemntfs.fname("proc"))
            do("mount --bind /sys %s" % imagemntfs.fname("sys"))

            do('mkdir -p "%s"' % imagemntfs.fname("boot/grub"))

            devmap = open(imagemntfs.fname("boot/grub/device.map"), "w")
            devmap.write("(hd0) %s\n" % poopdev)
            devmap.close()

            chroot(imagemnt, "update-initramfs -u -k all")

            # Replace groot and kopt because else they will be given
            # bad values
            #
            # FIXME - Pylint says: Using possibly undefined loop
            # variable 'entry' (undefined-loop-variable).  entry is
            # defined in the previous for-loop.
            do(r'chroot %s sed -in "s/^# groot=.*$/# groot=\(hd0,%d\)/" %s' %
               (imagemnt, int(entry.partnum) - 1, "/boot/grub/menu.lst"))
            do(r'chroot %s sed -in "s/^# kopt=.*$/# kopt=root=LABEL=%s/" %s' %
               (imagemnt, entry.label, "/boot/grub/menu.lst"))
--> menu.lst is modified, using last entry of fstab and not entry in partitions marked by "grub-install"

bgermann, 2021-04-13:
[Patch available](https://lists.linutronix.de/pipermail/elbe-devel/2021-April/005299.html)

closed 2021-04-30 by bgermann

bgermann, 2021-05-19:
This is contained in v13.3. Please reopen if it does not work for you.