Linutronix/elbe #297: In elbe v13.2 when using noauth setting pyhton complains about not initialized variable keyring

mrelyb, 2021-03-25:
When building image with "noauth" setting in project set, python complains about an uninitialized variable "keyring"

        if not self.xml.is_cross(host_arch):
            if self.xml.has("project/noauth"):        --> variable "keyring" not set in this branch
                cmd = '%s --no-check-gpg --arch=%s "%s" "%s" "%s"' % (
                    strapcmd, arch, suite, self.rfs.path, primary_mirror)
            else:
                if self.xml.has("project/mirror/cdrom"):
                    keyring = ' --keyring="%s"' % (
                        self.rfs.fname('/elbe.keyring'))
                else:
                    keyring = ''

                cmd = '%s --arch=%s %s "%s" "%s" "%s"' % (
                    strapcmd, arch, keyring, suite, self.rfs.path, primary_mirror)

            try:
                self.cdrom_mount()
                if keyring:                     ---> ERROR
                    do('apt-key --keyring "%s" add "%s/targetrepo/repo.pub"' % (
                        self.rfs.fname('/elbe.keyring'),
                        self.rfs.fname("cdrom")))
                do(cmd)

bgermann, 2021-03-25:
[Patch available](https://lists.linutronix.de/pipermail/elbe-devel/2021-March/005263.html).

mrelyb, 2021-03-25:
Patch works, thanks!

commit 49eaa69f (2021-04-01): Prevent uninitialized variable

With 990f94cf76 (rfs: fix cdrom build keyring problems), an uninitialized
variable case for 'keyring' was introduced when using noauth setting.

Fixes #297.

Reviewed-by: Torben Hohn <torben.hohn@linutronix.de>
closed 2021-04-01 by torbenh3