Linutronix/elbe #299: General question about license

laoshaw, 2021-04-28:
ELBE reminds of embedded debian project in the past, this project deserves more publicity and is very interesting for those who does not like Yocto's over-engineering design and long-hours rebuild even with sstate. Except for one thing, the GPL-3.0 license, what does it mean? does it mean all the elbe boilerplate scripts are GPL-3.0 but nothing else are impacted, or anything made with it may need to comply with GPL-3.0? I am not a lawyer but any projects chosen GPL-3.0 does not encourage people to join as much as it could be it seems. For example Yocto uses MIT license.

oldiob, 2021-04-28:
The source code is GPL-3.0.  I don't think that the output generated, i.e. the root file system, is.  This is why a licence file is generated by Elbe. Looking at other projects that are GPL-3.0, such as GCC or GNU Bison, theirs outputs are not GPL-3.0.  Now that's my understanding, but I would like to heard from the maintainer.

> any projects chosen GPL-3.0 does not encourage people to join

I disagree. Different peoples will contribute to the project that's all. In the end, if the outputs produced by Elbe is not affected by its license, then it doesn't affect its users.

bgermann, 2021-04-28:
Olivier is generally right. In the initvm, which can be seen as an intermediate build output, there is a lot of GPL-3 software installed (including Elbe itself) for which you have to consider when you distribute the initvm. But Elbe does not impose GPL-3 on the target output.

There is at least one Elbe build output that comes from Elbe and has a specific license on it, which is the GPL-2.0-only licensed toolchain-shar-extract.sh in an Elbe SDK. It is generated from elbepack/makofiles/toolchain-shar-extract.sh.mako. If you do not build an SDK (default), this is not applicable.

Be aware that Debian has some essential packages that are distributed under GPL-3.0 that will end up in your target unless you use the setsel, diet, or tighten modes. Most prominently bash and coreutils.

laoshaw, 2021-04-28:
Thanks. building my first image now.

closed 2021-04-28 by laoshaw