Linutronix/elbe #300: kvm vs lxc

laoshaw, 2021-04-28:
I use ubuntu 20.04 as development machine then run a debian 10 in virtualbox which does not do nested vt so can not build elbe on the virtualbox-debian based on my quick check. will LXC be possible as an option to build ELBE as there are probably more developers using Ubuntu than Debian these days.

For cross-compiling, is ELBE building from source or just pulling in various binary deb packaged from say ARM repos and integrate them into an image on x86 hosts? and if not rebuilding, what if I want to pass some buildflags to rebuild/make existing debs slightly differently, say -Os  for smaller size.

bgermann, 2021-04-29:
Why don't you install elbe on Ubuntu directly?

laoshaw, 2021-04-29:
from quick read on the docs it seems elbe is for debian 10 host only? I'm trying it on ubuntu 20.04 now. thanks!

bgermann, 2021-04-29:
Debian is the only officially supported distro but as Ubuntu should behave like it in most aspects, you can use it as well. Most of the processing will take place on the initvm anyway.

Elbe generally just pulls the binary packages but it can also build Debian source packages (cross or via QEMU userspace emulation) via pbuilder.

closed 2021-04-29 by bgermann