Linutronix/elbe #304: Build without initvm

thedavesmith, 2021-06-03:
Hi,

I would like to use elbe in an automated build system. Our automated build systems run on AWS infrastructure which doesn't allow for nested virtualization. In the video on the website, you mention that you can use elbe without the virtual machine. Can anyone please explain how to run a build without using initvm? I understand the drawbacks of not using initvm but at this point I can't see any other option.

Thanks in advance.

bgermann, 2021-07-09:
Start the daemon and let the client connect to the right ELBE_SOAPPORT (7588 instead of the default 7587).

closed 2021-08-12 by bgermann

jogness, 2021-10-21:
When you run "elbe initvm create" it will create a hard drive image for the initvm (called initvm.img). This is just an x86_64 hard drive image. If you use this for your root filesystem in your AWS instance, it should work fine. There is nothing special about the initvm virtual machine.

fmg-dave, 2021-10-21:
@dbraunwarth 

I have spent some time over the last few weeks getting ELBE to work for my use case. I need a 100% repeatable build and I wanted to use GitHub Actions to do it. GitHub Hosted Runners are not immutable. They are constantly "improved" with packages being added, removed and updated. This means re-running an old workflow may produce a different result. This can be addressed by running your own self hosted runner (on AWS for example) and tightly controlling it or scripting its creation (think ansible). My preference is to provision a clean environment for each build. GitHub hosted runners do this by provisioning a new runner VM automatically for each job. You can do this in AWS as well but it requires a bit of work and the initvm would need to be built each time (or maybe copied) and this takes significant time. The way to avoid this is to rely less on the runner by making sure all your Jobs are run within immutable containers. My Jobs run within a container based off the  `debian:bullseye-20211011-slim` image. If I run the job within the container, I care less about the runner host.

Now if I were to run ELBE as intended, with an initvm, I would have excessive layers of virtualisation. It would be: Bare Metal Hardware -> Hypervisor -> Runner Virtual Machine -> Docker Container - > initvm.

So to address this I made my container basically act as the initvm. For reproducability puposes, I maintain my own private Debian mirror which has snapshot capabilities. I point container apt sources to my mirror snapshots so I can always retrieve the same version of packages. I use these sources to install all of the dependencies for ELBE. Within the container I also download and add a specific version of ELBE. I use the SHA hash for the commit I want, download the zip and add the elbe binary to $PATH. At this point I now have a container that has the capability to run the elbe daemon. The typical way to interact with ELBE is to use `elbe initvm submit`. This command fires off a series of commands to the ELBE daemon within the initvm. I have written a simple script to perform similar a function. It first starts the ELBE daemon in the container and then fires off the same commands.

I realise the container uses the kernel from the host but that is sufficient for me and should have minimal, if any, effect on the build reproducability.

I am still working through this. The following is not a complete working example (I've had to strip out many things that are not relevant to this conversation) but it gives some snippets and shows the gist of the approach.

Dockerfile:

```dockerfile
FROM debian:bullseye-20210621-slim

# This is where I set the sources to my private mirror and update any packages already in the container to be version synced with my mirror.

# Install ELBE dependencies.
RUN DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends -y \
  # python3-elbe-bin
  python3 python3-mako debian-archive-keyring \
  # python3-elbe-common
  python3 python3-lxml python3-apt python3-gpg python3-libvirt wget cpio \
  # elbe
  qemu-utils qemu-kvm p7zip-full make lsb-release \
  # python3-elbe-daemon
  python3 python3-cherrypy3 python3-beaker lsb-base \
  # python3-elbe-buildenv
  python3 apt-rdepends binfmt-support debootstrap dosfstools dpkg-dev genisoimage mtd-utils python3-apt python3-junit.xml python3-mako python3-parted python3-passlib python3-sqlalchemy python3-debian qemu-user-static qemu-utils reprepro rsync kpartx squashfs-tools sudo pbuilder git ca-certificates subversion haveged devscripts \
  # python3-elbe-soap
  python3 python3-spyne \
  # python3-elbe-control
  python3 python3-suds python3-debian libvirt-clients libvirt-daemon-system

# This is using a prerelease until bullseye is supported in the next release.
ENV PATH="/opt/elbe:${PATH}"
RUN wget https://github.com/Linutronix/elbe/archive/c931652b2bd967c3069d80750dcf10bf79997f4b.tar.gz \
&& tar -xf c931652b2bd967c3069d80750dcf10bf79997f4b.tar.gz -C /opt \
&& rm c931652b2bd967c3069d80750dcf10bf79997f4b.tar.gz \
&& mv /opt/elbe-c931652b2bd967c3069d80750dcf10bf79997f4b /opt/elbe \
&& elbe db init

# Additions required for not running an elbe initvm. This is a dummy source to trick ELBE into thinking it is an initvm.
COPY elbe/source.xml /var/cache/elbe/source.xml

COPY elbe/build_elbe_project.sh /opt/elbe/build_elbe_project.sh

# Fix for elbe. grub-install in bullseye changed the default target, we need to fix this until fixed upstream.
# See: https://github.com/Linutronix/elbe/pull/312
RUN sed -i "s/grub-install --no-floppy/grub-install --target=i386-pc --no-floppy/" /opt/elbe/elbepack/hdimg.py
```

build_elbe_project:

```shell
#!/bin/bash

set -e

# A whole bunch of setup is done here

...

# Look for a "files" directory alongside the XML files and add its contents to the project XML file.
if [ -d "$BUILD_DIR/files" ]; then
    echo "Adding files to project XML..."
    tar -cjf "$BUILD_DIR/files.tar.bz2" -C "$BUILD_DIR/files" .
    elbe chg_archive "$BUILD_DIR/$XML_FILE" "$BUILD_DIR/files.tar.bz2"
[... 4 lines cut ...]

echo "Starting the elbe daemon..."
elbe daemon --soap &

echo "Creating an elbe project..."
ELBE_PROJECT_ID=$(elbe control create_project)

echo "Adding the XML file to the project..."
elbe control set_xml "$ELBE_PROJECT_ID" "$BUILD_DIR/processed.xml"

echo "Building the project..."
elbe control build "$ELBE_PROJECT_ID"

echo "Wait for the project to complete..."
elbe control wait_busy "$ELBE_PROJECT_ID"

[... cut ...]
**Note: You have to run the container in privileged mode.** This is due to ELBE creating loop devices during a build.

This may be a different direction to the original question of running on AWS but hopefully this give you some ideas.

Overall I think the docker approach is much more flexible and suitable to CI systems that have ephemeral build environments. I would like to see this capability added to ELBE as a first class citizen. If the maintainers of ELBE are open to this approach I would be happy to contribute the bulk of the work.