Linutronix/elbe #306: binary not working with gpthd

martinsteih, 2021-08-20:
I tried to insert a binary command within gpthd, but XML validation tells me it's an unexpected command, though it works within msdoshd.

bgermann, 2021-08-25:
That is right, it is missing in the schema for GPT. Would that be a correct use case?

martinsteih, 2021-08-25:
Yes, that’s my use case. I just want to copy my u-Boot binary to a specific offset of my first partition. Exactly like the example with msdoshd except that I need gpthd. 
Curiously, I did add it to the Schema but Elbe still complained with „unexpected …“ but I‘ll double check it once again.

martinsteih, 2021-09-01:
I copied the binary part of msdoshd to gpthd in the schema, but still, it does not work. Within the code, I'm unable to find anything that might not work, so I am a little confused.

bgermann, 2021-09-01:
If you have elbe installed, the schema is not taken from git but from the system's xml catalogue.

martinsteih, 2021-09-01:
Still the same. I replaced every occurrence of the schema and even uninstalled elbe to make sure my repo clone is the only possible source of anything. Still, I get the same error. I even double-checked the file itself. As far as I see it, there is only one occurrence of gphd that is to be edited. Next, to be absolutely certain, I removed my initvm and created a new one. But the error still persists. Any hints?

bgermann, 2021-09-01:
Please use --devel on creating the initvm so that the file is used there as well.

martinsteih, 2021-09-01:
Ok, thank you so much. That did the trick. Now it runs and builds the image.

commit f6538da3 (2021-09-08): schema: Support <binary> for gpthd

<binary> support is implemented for all partition formats but the
schema does not allow <binary> in <gpthd>.

Allow it for gpthd as well.

Closes #306.

Reviewed-by: Torben Hohn <torben.hohn@linutronix.de>
closed 2021-09-01 by bgermann