Linutronix/elbe #324: elbe preprocess failes due to artificial? limit in https://github.com/Linutronix/elbe/blob/releases/v13.2/schema/dbsfed.xsd

danielbisar, 2022-03-07:
Steps to reproduce:

- create root_fs.xml
- add more than 10 repos via <url> to <url-list>

Result: Error message:
`... elbe_rootfs.xml:418 error Element 'url': This element is not expected.`

Expected behavior:
1. better error message
2. even better: remove the maxOccurs="10" from the url element under xml-list from the schema

Is this expected behavior? What shall I do if i need more repos?

manut, 2022-03-21:
> Closed #324 via b7911ec.

The url-list is also used for preseeding the Debian installer during Initvm creation:

https://github.com/Linutronix/elbe/blob/4d0f93be48402e36a040824f9d7d1d72c984e544/elbepack/init/preseed.cfg.mako#L83

And there is a limit:

https://www.debian.org/releases/testing/amd64/apbs04.en.html#preseed-apt

This is why this was limited once..

Manuel

bgermann, 2022-03-21:
Thanks for pointing this out!

danielbisar, 2022-03-21:
Thanks for fixing :)

bgermann, 2022-03-21:
I think I will revert it again (see @manut's comment).

danielbisar, 2022-03-22:
Wouldn't it be possible to use another value for the seeding? I mean it would just need to use another xsd oder update the xml-node of the schema according to the actual limit. As far as i could see the limit mentioned by @manut is system dependend, so there is no actual hard limit to 10. And to make elbe fail when not even using seeding doesn't make sense to me.

commit b7911ec7 (2022-03-21): schema: remove url-list entries limit

This patch fixes #324.

Reviewed-by: Bastian Germann <bage@linutronix.de>