Linutronix/elbe #325: Boot problem

SerMel13, 2022-03-11:
Hello,
Could you help me with unbootable elbe image? 

I created elbe image via "initvm submit ...". There were no error messages. Then I wrote the result sda.img.tar.gz to sda of VM (via tar xf0z and dd).

The boot process hangs before switching root without any warnings or errors (on VirtualBox and Qemu-KVM). 

The grub command and partitions layout seems to be correct. Is there any simple way to turn on "debug mode in initrd" for elbe images (without patching initrd file and some other tricks)?

SerMel13, 2022-03-11:
[image]

bgermann, 2022-03-11:
How are there no errors? There are plenty of them. Most probably you have a GPT layout problem.

SerMel13, 2022-03-11:
[image]

SerMel13, 2022-03-11:
Ran parted from LIveCD and remove DVD from VM.
Still hangling.

SerMel13, 2022-05-19:
Sorry for not updating. Adding devtmpfs.mount=1 to the linux command in grub config helped.

closed 2022-05-19 by bgermann