Linutronix/elbe #336: SDK Relocation And Scripts

FedericoVaga, 2022-05-19:
I see in the `elbepack/makofiles/toolchain-shar-extract.sh.mako` file that during relocation, the script is going to fix the interpreter and the rpath on a number of binary files.

Then for ASCII `.so` the script is going to fix internal paths according to the final SDK location. I'm wondering if this part shouldn't be extended to scripts as well? For example, many scripts have this:

    #!/bin/bash
    #!/bin/python
    #!/bin/env bash
    ...

This means that if executed from the SDK they will use the host interpreter and not the SDK one.

Is this expected to work like that?

manut, 2022-05-19:
Hi,

since the environment Setup script prefixes the path with the host sysroot the proper interpreter should be used.

Manuel

FedericoVaga, 2022-05-19:
I'm referring to other parts of the SDK sysroot, not just the ones related to building C/C++ code.

If, in the SDK I installed some other tools (let's say python) that I want to use for developing, shouldn't these tools call `/path-sdk/usr/bin/python` instead `/usr/bin/python`?

If this is not supposed to be the use case, why then not slimming down the SDK and removing whatever does not contribute to build C/C++ code?

I'm naively asking, I'm not requesting any feature nor pretending that what I'm trying is good :smiley:

manut, 2022-10-11:
It should be also okay for interpreters:

export PATH="${sdk_path}/sysroots/host/usr/bin:${sdk_path}/sysroots/host/usr/sbin:${sdk_path}/sysroots/host/bin:${sdk_path}/sysroots/host/sbin:$PATH"

In the environment Setup script ensures that the interpreters available in the host sysroot are preferred to the ones in the system

Manuel

FedericoVaga, 2022-10-11:
What you wrote is okay if the script uses

```
#!/usr/bin/env python
```

But if the path to the interpreter is hardcoded into the script, then it will not use PATH. For example:

```
#!/usr/bin/python
```

FedericoVaga, 2022-11-14:
On the same topic (SDK relocation) but a different issue. For example, let's take the file `sysroots/host/lib/x86_64-linux-gnu/libc.so` (there are others)

```
/* GNU ld script
   Use the shared library, but some functions are only in
   the static library, so try that secondarily.  */
OUTPUT_FORMAT(elf64-x86-64)
GROUP ( /lib/x86_64-linux-gnu/libc.so.6 <relocation-path>/sysroots/host/usr/lib/x86_64-linux-gnu/libc_nonshared.a  AS_NEEDED ( /lib64/ld-linux-x86-64.so.2 ) )
```

In this file, the only path that get fixed is the one containing `/usr/lib`, the others located in `/lib` and `/lib64` do not get update. Therefore, they rely on the SDK host system.

Doesn't the following line need an update? It looks like not enough to replace paths behind `/usr`

 https://github.com/Linutronix/elbe/blob/fe110e8be6b8301103b50c441d6bab549bcb2447/elbepack/makofiles/toolchain-shar-extract.sh.mako#L366

bgermann, 2023-04-24:
Yes, please propose a patch.

FedericoVaga, 2023-05-16:
> You use the ` initvm --devel create` option.

Thanks, I did not notice that obvious option :D

FedericoVaga, 2023-05-26:
Fixed in: https://github.com/Linutronix/elbe/pull/378

closed 2023-05-26 by FedericoVaga