Linutronix/elbe #342: Elbe Docs not available/empty

martinbu, 2022-07-26:
I have noticed that the content of [Elbe Docs](https://elbe-rfs.org/docs/sphinx/index.html) has been empty for a while. Maybe since the 14.5 release?

Does something went wrong here?

bgermann, 2022-07-26:
That is a known issue. I am working on it.

bgermann, 2022-08-25:
The info on the sphinx generated website is also available in git.

bgermann, 2022-09-22:
I have reverted the docs to the last known-good version.

closed 2022-09-22 by bgermann

d4nuu8, 2022-11-03:
> I have reverted the docs to the last known-good version.

I think this is still a issue. There is only the man-page of `elbe-add` available:
[image]

bgermann, 2022-11-03:
Yes, I know. It will be rectified with the next version. Please just switch to a different version on the side bar for now.