Linutronix/elbe #346: Syntax Error during ELBE Docker build

mrelyb, 2022-10-13:
While building ELBE docker image with version 14.6 the following syntax error occurs:

Setting up elbe-schema (14.6.1) ...
Setting up python3-elbe-common (14.6.1) ...
  File "/usr/lib/python3/dist-packages/elbepack/repodir.py", line 40
    bin_el.text = f"http://LOCALMACHINE:{httpd.server_address[1]} {repo[1]}"
                                                                           ^
SyntaxError: invalid syntax

dpkg: error processing package python3-elbe-common (--configure):
 subprocess installed post-installation script returned error exit status 1
dpkg: dependency problems prevent configuration of python3-elbe-control:
 python3-elbe-control depends on python3-elbe-common (= 14.6.1); however:
  Package python3-elbe-common is not configured yet.

dpkg: error processing package python3-elbe-control (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of python3-elbe-bin:
 python3-elbe-bin depends on python3-elbe-common (= 14.6.1); however:
  Package python3-elbe-common is not configured yet.

dpkg: error processing package python3-elbe-bin (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of elbe:
 elbe depends on python3-elbe-bin (= 14.6.1); however:
  Package python3-elbe-bin is not configured yet.
 elbe depends on python3-elbe-control (= 14.6.1); however:
  Package python3-elbe-control is not configured yet.

dpkg: error processing package elbe (--configure):
 dependency problems - leaving unconfigured

bgermann, 2022-10-13:
README: "Please note that we currently target Python 3.6+"
It is not a syntax error in Python 3.6+

closed 2022-10-13 by bgermann

mrelyb, 2022-10-13:
So what should I do to make the Docker image under contrib/dockerfile build with 14.6?
Replace 
# This Dockefile generate a image for the elbe buildsystem
FROM elbeproject/debian-stretch:latest

by 

# This Dockefile generate a image for the elbe buildsystem
FROM elbeproject/debian-bullseye:latest
?

bgermann, 2022-10-13:
Right.

bgermann, 2022-10-13:
I did not read properly. You would have to use the official Debian images (debian:bullseye).

mrelyb, 2022-10-13:
Yes I supposed that. Thanks for your help

bgermann, 2022-10-13:
See https://lists.linutronix.de/pipermail/elbe-devel/2022-October/005981.html