Linutronix/elbe #347: Starting the initvm Failed

xliang76, 2022-10-15:
I tried to create the image with x86_64-pc-hdimage-gnome3.xml from example.
I got error 
ibvirt: QEMU Driver error : internal error: qemu unexpectedly closed the monitor: 2022-10-15T17:26:06.524883Z qemu-system-x86_64: The -accel and "-machine accel=" options are incompatible
any idea?

bgermann, 2022-10-15:
Your host system's libvirt and qemu versions have to be compatible.

xliang76, 2022-10-15:
I am running from Ubuntu 22.04 and elbe is checkout from this repo.
How do I check if the libvirt and qemu version is compatible?
Thank you

bgermann, 2022-10-15:
which versions are they? Any backports/third party repos (except for elbe?

xliang76, 2022-10-15:
I didn't add third party repos as it was fresh Ubuntu install.
And the libvirt is 8.0.0 The qemu is 6.2.0

bgermann, 2022-10-15:
Hm... The commit that changed things is at https://gitlab.com/qemu-project/qemu/-/commit/976e8c54142586285f71f524308e4715f6e04e2d at the qemu side. But that is quite old.
Please explain the other question in more detail a new issue.

xliang76, 2022-10-15:
thank you for this. I tried to track down the function call, it run into the libvirt. I can't find it in my system. So, it must be built one. Looks like it need to install other version or manually update it. 
For the second question, it is not issue. It is a learning question.

bgermann, 2023-02-06:
This was observed on bookworm, so I am reopening.

commit 52526564 (2023-02-23): Replace qemu -enable-kvm option

QEMU's -enable-kvm option maps to the old-style -machine accel which is
incompatible with the -accel option in current qemu versions. The -accel
option is implied by libvirt's 'hvm' machine type that is used with Elbe's
defaults.

Replace the -enable-kvm that ends up in libvirt so Elbe is compatible with
bookworm. The replacement is at least available since buster (oldest
supported release for the client).

Fixes: #347

Reviewed-by: Kurt Kanzenbach <kurt@linutronix.de>
closed 2023-02-23 by bgermann