Linutronix/elbe #348: Two UEFI folders after build (/boot/efi/EFI/BOOT/ and /boot/efi/EFI/debian/)

koberbe-jh, 2022-10-20:
Hello everyone,
I stumbled upon the fact that ELBE creates two folders with EFI binaries - /boot/efi/EFI/BOOT/ and /boot/efi/EFI/debian/ - by copying the BOOT folder to the debian folder:
https://github.com/Linutronix/elbe/blob/0f94dee59e7984bc8fb1bc0d018df021b7c9d5c3/elbepack/hdimg.py#L206
What I also found is the /boot/efi/EFI/BOOT/ seems to be some kind of default target folder for efi binaries when using grub-install. What I don't understand is: What do we need /boot/efi/EFI/BOOT/ for if it was already copied? Can it be removed? Does anyone have some tips or documentation about this?

bgermann, 2022-10-20:
Yes, I think to remember it to be the default target when you are starting the system from a "removable media". If all you do is starting it from HDD you can remove it. The problem why this is done manually is that grub-install has runtime checks on it being run on EFI or BIOS and behaving differently. See also the commit that introduced this at a247a279569aee71bc65c255ea440748e67c107d.

koberbe-jh, 2022-10-20:
Thanks for the quick answer. That BOOT is used from removable media is a very interesting information. I think I now understand the motivation.

closed 2022-10-20 by koberbe-jh