Linutronix/elbe #353: How to ignore missing license file in packages?

coveritytest, 2022-11-30:
How can I ignore missing license files in packages, so that the build does not fail:

```
[ERROR] Error while processing license file /var/cache/elbe/9f1f5d75-b96c-4465-a50c-a6b790d8f1a8/chroot/usr/share/doc/some-package/copyright
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/elbepack/efilesystem.py", line 203, in write_licenses
    with io.open(copyright_fname, "r",
FileNotFoundError: [Errno 2] No such file or directory: '/var/cache/elbe/9f1f5d75-b96c-4465-a50c-a6b790d8f1a8/chroot/usr/share/doc/some-package/copyright'
```

bgermann, 2022-11-30:
You add the required copyright file to that package.

closed 2022-11-30 by bgermann

coveritytest, 2022-11-30:
If it was my package sure, but if those packages are not under my control it's a problem. And it should not be an error rather a warning as it's not critical.

bgermann, 2022-11-30:
It is critical because this is a very essential file. If you give a hint which package this is I can maybe help.

coveritytest, 2022-11-30:
The cuda suite has a bunch of packages with missing copyright files, for example the following packages:

```
cuda
cuda-11-6
cuda-command-line-tools-11-6
cuda-compiler-11-6
cuda-drivers
cuda-drivers-510
cuda-libraries-11-6
cuda-libraries-dev-11-6
cuda-nsight-compute-11-6
cuda-nsight-systems-11-6
cuda-runtime-11-6
cuda-toolkit-11-6
cuda-tools-11-6
cuda-visual-tools-11-6
libcufile-dev-11-6
nsight-systems-2021.5.2
```

bgermann, 2022-11-30:
Why don't you use https://tracker.debian.org/pkg/nvidia-cuda-toolkit?
Alternatively, please post a patch for this that makes this a warning.

coveritytest, 2022-11-30:
Basically because a certain version is needed. Will post a patch.

coveritytest, 2022-12-01:
Do you want an attribute in <pkg-list> to mark packages where we want to skip the license file or just write no license file at all, if none is found?

bgermann, 2022-12-01:
skip the copyright file and put a warning that it did not exist.

coveritytest, 2022-12-01:
Would you please consider to bump elbe version to 14.6.2 after the last commit? This would help us to use this (for us) important feature with Debian installations.

bgermann, 2022-12-01:
There is more involved in the release process and I would like to get some more stuff in. Do not expect a release before 15 Dec.

coveritytest, 2022-12-01:
Okay, will use master so long.