Linutronix/elbe #356: GPG key has been expired

stmork, 2022-12-12:
Unfortunately the GPG key has been expired:
```
gpg elbe-archive-keyring.gpg 
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
pub   rsa2048 2012-11-23 [SC] [expired: 2022-12-10]
      10E0A5355609F7C6A55A17F636AA35FF22BB8F84
uid           ELBE Devel (Linutronix Elbe developers) <elbe-devel@linutronix.de>
sub   rsa2048 2012-11-23 [E] [expired: 2017-12-11]
```

bgermann, 2022-12-12:
Then please update it from where you got it...

closed 2022-12-12 by bgermann

stmork, 2022-12-12:
I've updated the key right now:
`wget -O elbe-archive-keyring.gpg --no-proxy http://debian.linutronix.de/elbe/elbe-repo.pub.gpg`

bgermann, 2022-12-13:
Should be up to date now. For the record: There is a Debian package elbe-archive-keyring available that also has it.

stmork, 2022-12-14:
Thank you. :-)

Meanwhile I found the package, too, but an `elbe initvm create` needed manual work in the busybox, since I assume the creation scripts download the key directly. The package verification disabled the elbe repositories in _sources.list_ so an installation didn't succeed. After manually installing the elbe-archive-keyring package, updating the key in _/etc/apt/trusted.gpg.d_ and reenabling the _sources.list_ the installation completed successfully.