Linutronix/elbe #360: How to extract first partition only in <gpthd>?

coveritytest, 2023-01-20:
I need to extract the first sectors of sda and the partition sda1 so I can boot the rfs from my hard drive on x86_64. While it works fine with `examples/x86_64-pc-hdimg-grub-uefi-buster.xml` I do not want to extract sda2 as it is too big. So instead of the example:

```
<images>
	<!-- use GPT for UEFI -->
	<gpthd>
		<name>sda.img</name>
		<size>2GiB</size>
		<!-- install grub -->
		<grub-install/>
		<!-- the EFI system partition -->
		<partition> 
			<size>200MiB</size>
			<label>uefi</label>
			<bootable>true</bootable>
		</partition>
		<partition>
			<size>remain</size>
			<label>rfs</label>
		</partition>
	</gpthd>
</images>

<project-finetuning>
	<losetup img="sda.img">
		<extract_partition part="1">sda1.img</extract_partition>
		<extract_partition part="2">sda2.img</extract_partition>
		<command part="1">
			ls $ELBE_MNT
		</command>
	</losetup>
	<set_packer packer="none">sda1.img</set_packer>
	<set_packer packer="tarxz">sda2.img</set_packer>
</project-finetuning>
```

I tried to omit the rfs on sda2:

```
<images>
	<!-- use GPT for UEFI -->
	<gpthd>
		<name>sda.img</name>
		<size>1GiB</size>
		<!-- install grub -->
		<grub-install/>
		<!-- the EFI system partition -->
		<partition> 
			<size>200MiB</size>
			<label>uefi</label>
			<bootable>true</bootable>
		</partition>
	</gpthd>
</images>

<project-finetuning>
	<losetup img="sda.img">
		<extract_partition part="1">sda1.img</extract_partition>
		<command part="1">
			ls $ELBE_MNT
		</command>
	</losetup>
	<set_packer packer="none">sda1.img</set_packer>
</project-finetuning>

```
While the first sectors on sda.img look fine, sda1.img is only formatted but EFI/BOOT is not on it. What am I doing wrong here?

bgermann, 2023-01-20:
Without using fstab you will not get any data on the partitions.

closed 2023-01-20 by bgermann

coveritytest, 2023-01-24:
Unfortunately not solved yet. The fstab is not the problem, I just did not pasted it above. The problem is that when I remove the rfs from the <gpthd> and <project-finetuning>, sda1.img does not contain /boot/efi. Here is my complete XML:

```
<!--
SPDX-License-Identifier: 0BSD
SPDX-FileCopyrightText: Linutronix GmbH
-->
<ns0:RootFileSystem xmlns:ns0="https://www.linutronix.de/projects/Elbe" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" created="2009-05-20T08:50:56" revision="6" xsi:schemaLocation="https://www.linutronix.de/projects/Elbe dbsfed.xsd">
	<project>
		<name>amd64-grub-uefi-buster</name>
		<version>1.0</version>
		<description>
			verify successful UEFI boot by checking
			lsmod | grep efi
			ls /sys/firmware/efi

			test UEFI boot in qemu by installing "ovmf" and using the parameter
			-bios /usr/share/qemu/OVMF.fd
		</description>
		<buildtype>amd64</buildtype>
		<mirror>
			<primary_host>ftp.de.debian.org</primary_host>
			<primary_path>/debian</primary_path>
			<primary_proto>http</primary_proto>
			<url-list>
				<url>
					<binary>http://security.debian.org buster/updates main</binary>
					<source>http://security.debian.org buster/updates main</source>
				</url>
			</url-list>
		</mirror>
		<suite>buster</suite>
	</project>
	<target>
		<hostname>lxvm</hostname>
		<domain>tec.linutronix.de</domain>
		<passwd>foo</passwd>
		<console>ttyS0,115200</console>
		<package>
			<tar>
				<name>grub-uefi-rfs.tgz</name>
			</tar>
		</package>
		<images>
			<!-- use GPT for UEFI -->
			<gpthd>
				<name>sda.img</name>
				<size>2GiB</size>
				<!-- install grub -->
				<grub-install/>
				<!-- the EFI system partition -->
				<partition> 
					<size>200MiB</size>
					<label>uefi</label>
					<bootable>true</bootable>
				</partition>
			</gpthd>
		</images>
		<fstab>
			<!-- the EFI system partition needs to be mounted at /boot/efi -->
			<bylabel>
				<label>uefi</label>
				<mountpoint>/boot/efi</mountpoint>
				<fs>
					<type>vfat</type>
				</fs>
			</bylabel>
		</fstab>
		<install-recommends />
		<pkg-list>
			<!-- install the signed EFI version of grub -->
			<pkg>grub-efi-amd64-signed</pkg>
			<pkg>shim-signed</pkg>
			<pkg>linux-image-amd64</pkg>
		</pkg-list>
		<project-finetuning>
			<losetup img="sda.img">
				<extract_partition part="1">sda1.img</extract_partition>
				<command part="1">
				  ls $ELBE_MNT
				</command>
			</losetup>
			<set_packer packer="none">sda1.img</set_packer>
		</project-finetuning>
	</target>
</ns0:RootFileSystem>
```
And here is the patch compared to examples/x86_64-pc-hdimg-grub-uefi-buster.xml:

```
--- <unnamed>
+++ <unnamed>
@@ -51,21 +51,9 @@
 					<label>uefi</label>
 					<bootable>true</bootable>
 				</partition>
-				<partition>
-					<size>remain</size>
-					<label>rfs</label>
-				</partition>
 			</gpthd>
 		</images>
 		<fstab>
-			<bylabel>
-				<label>rfs</label>
-				<mountpoint>/</mountpoint>
-				<fs>
-					<type>ext4</type>
-					<tune2fs>-i 0</tune2fs>
-				</fs>
-			</bylabel>
 			<!-- the EFI system partition needs to be mounted at /boot/efi -->
 			<bylabel>
 				<label>uefi</label>
@@ -85,13 +73,11 @@
 		<project-finetuning>
 			<losetup img="sda.img">
 				<extract_partition part="1">sda1.img</extract_partition>
-				<extract_partition part="2">sda2.img</extract_partition>
 				<command part="1">
 				  ls $ELBE_MNT
 				</command>
 			</losetup>
 			<set_packer packer="none">sda1.img</set_packer>
-			<set_packer packer="tarxz">sda2.img</set_packer>
 		</project-finetuning>
 	</target>
 </ns0:RootFileSystem>
```
I don't understand why /boot/efi is not contained on sda1.img.