Linutronix/elbe #361: How to extract sda1.img containing /boot/efi?

coveritytest, 2023-01-24:
I need to extract sda1.img containing /boot/efi. But unfortunately sda1.img does not contain /boot/efi, it only contains an empty fat formatted image. Here is my complete XML:

```
<!--
SPDX-License-Identifier: 0BSD
SPDX-FileCopyrightText: Linutronix GmbH
-->
<ns0:RootFileSystem xmlns:ns0="https://www.linutronix.de/projects/Elbe" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" created="2009-05-20T08:50:56" revision="6" xsi:schemaLocation="https://www.linutronix.de/projects/Elbe dbsfed.xsd">
	<project>
		<name>amd64-grub-uefi-buster</name>
		<version>1.0</version>
		<description>
			verify successful UEFI boot by checking
			lsmod | grep efi
			ls /sys/firmware/efi

			test UEFI boot in qemu by installing "ovmf" and using the parameter
			-bios /usr/share/qemu/OVMF.fd
		</description>
		<buildtype>amd64</buildtype>
		<mirror>
			<primary_host>ftp.de.debian.org</primary_host>
			<primary_path>/debian</primary_path>
			<primary_proto>http</primary_proto>
			<url-list>
				<url>
					<binary>http://security.debian.org buster/updates main</binary>
					<source>http://security.debian.org buster/updates main</source>
				</url>
			</url-list>
		</mirror>
		<suite>buster</suite>
	</project>
	<target>
		<hostname>lxvm</hostname>
		<domain>tec.linutronix.de</domain>
		<passwd>foo</passwd>
		<console>ttyS0,115200</console>
		<package>
			<tar>
				<name>grub-uefi-rfs.tgz</name>
			</tar>
		</package>
		<images>
			<!-- use GPT for UEFI -->
			<gpthd>
				<name>sda.img</name>
				<size>2GiB</size>
				<!-- install grub -->
				<grub-install/>
				<!-- the EFI system partition -->
				<partition> 
					<size>200MiB</size>
					<label>uefi</label>
					<bootable>true</bootable>
				</partition>
			</gpthd>
		</images>
		<fstab>
			<!-- the EFI system partition needs to be mounted at /boot/efi -->
			<bylabel>
				<label>uefi</label>
				<mountpoint>/boot/efi</mountpoint>
				<fs>
					<type>vfat</type>
				</fs>
			</bylabel>
		</fstab>
		<install-recommends />
		<pkg-list>
			<!-- install the signed EFI version of grub -->
			<pkg>grub-efi-amd64-signed</pkg>
			<pkg>shim-signed</pkg>
			<pkg>linux-image-amd64</pkg>
		</pkg-list>
		<project-finetuning>
			<losetup img="sda.img">
				<extract_partition part="1">sda1.img</extract_partition>
				<command part="1">
				  ls $ELBE_MNT
				</command>
			</losetup>
			<set_packer packer="none">sda1.img</set_packer>
		</project-finetuning>
	</target>
</ns0:RootFileSystem>
```
It is based on `examples/x86_64-pc-hdimg-grub-uefi-buster.xml`, I just removed the part where the rfs is extracted, which I do not want. Here is the patch:

```
--- <unnamed>
+++ <unnamed>
@@ -51,21 +51,9 @@
 					<label>uefi</label>
 					<bootable>true</bootable>
 				</partition>
-				<partition>
-					<size>remain</size>
-					<label>rfs</label>
-				</partition>
 			</gpthd>
 		</images>
 		<fstab>
-			<bylabel>
-				<label>rfs</label>
-				<mountpoint>/</mountpoint>
-				<fs>
-					<type>ext4</type>
-					<tune2fs>-i 0</tune2fs>
-				</fs>
-			</bylabel>
 			<!-- the EFI system partition needs to be mounted at /boot/efi -->
 			<bylabel>
 				<label>uefi</label>
@@ -85,13 +73,11 @@
 		<project-finetuning>
 			<losetup img="sda.img">
 				<extract_partition part="1">sda1.img</extract_partition>
-				<extract_partition part="2">sda2.img</extract_partition>
 				<command part="1">
 				  ls $ELBE_MNT
 				</command>
 			</losetup>
 			<set_packer packer="none">sda1.img</set_packer>
-			<set_packer packer="tarxz">sda2.img</set_packer>
 		</project-finetuning>
 	</target>
 </ns0:RootFileSystem>
```
Why is sda1.img empty and does not contain /boot/efi?

bgermann, 2023-01-24:
You removed mount pont /. Please have / and try again.

coveritytest, 2023-01-24:
I added the mountpoint again, problem still exists, sda1.img does not contain /boot/efi. Here's the adapted XML:

```
<!--
SPDX-License-Identifier: 0BSD
SPDX-FileCopyrightText: Linutronix GmbH
-->
<ns0:RootFileSystem xmlns:ns0="https://www.linutronix.de/projects/Elbe" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" created="2009-05-20T08:50:56" revision="6" xsi:schemaLocation="https://www.linutronix.de/projects/Elbe dbsfed.xsd">
	<project>
		<name>amd64-grub-uefi-buster</name>
		<version>1.0</version>
		<description>
			verify successful UEFI boot by checking
			lsmod | grep efi
			ls /sys/firmware/efi

			test UEFI boot in qemu by installing "ovmf" and using the parameter
			-bios /usr/share/qemu/OVMF.fd
		</description>
		<buildtype>amd64</buildtype>
		<mirror>
			<primary_host>ftp.de.debian.org</primary_host>
			<primary_path>/debian</primary_path>
			<primary_proto>http</primary_proto>
			<url-list>
				<url>
					<binary>http://security.debian.org buster/updates main</binary>
					<source>http://security.debian.org buster/updates main</source>
				</url>
			</url-list>
		</mirror>
		<suite>buster</suite>
	</project>
	<target>
		<hostname>lxvm</hostname>
		<domain>tec.linutronix.de</domain>
		<passwd>foo</passwd>
		<console>ttyS0,115200</console>
		<package>
			<tar>
				<name>grub-uefi-rfs.tgz</name>
			</tar>
		</package>
		<images>
			<!-- use GPT for UEFI -->
			<gpthd>
				<name>sda.img</name>
				<size>2GiB</size>
				<!-- install grub -->
				<grub-install/>
				<!-- the EFI system partition -->
				<partition> 
					<size>200MiB</size>
					<label>uefi</label>
					<bootable>true</bootable>
				</partition>
			</gpthd>
		</images>
		<fstab>
			<bylabel>
				<label>rfs</label>
				<mountpoint>/</mountpoint>
				<fs>
					<type>ext4</type>
					<tune2fs>-i 0</tune2fs>
				</fs>
			</bylabel>
			<!-- the EFI system partition needs to be mounted at /boot/efi -->
			<bylabel>
				<label>uefi</label>
				<mountpoint>/boot/efi</mountpoint>
				<fs>
					<type>vfat</type>
				</fs>
			</bylabel>
		</fstab>
		<install-recommends />
		<pkg-list>
			<!-- install the signed EFI version of grub -->
			<pkg>grub-efi-amd64-signed</pkg>
			<pkg>shim-signed</pkg>
			<pkg>linux-image-amd64</pkg>
		</pkg-list>
		<project-finetuning>
			<losetup img="sda.img">
				<extract_partition part="1">sda1.img</extract_partition>
				<command part="1">
				  ls $ELBE_MNT
				</command>
			</losetup>
			<set_packer packer="none">sda1.img</set_packer>
		</project-finetuning>
	</target>
</ns0:RootFileSystem>
```

bgermann, 2023-01-24:
You added the mountpoint but not the corresponding partition. Why don't you just build the untouched example and report back? This will make things a lot easier.

coveritytest, 2023-01-24:
The untouched example works fine, but I do not want sda2.img, as this partition/image would be huge. I only want sda1.img to be extracted. Isn't this possible?

bgermann, 2023-01-24:
You have to have / pointing to some partition when you create images containing filesystems. Currently, there is no way around that.

closed 2023-01-24 by bgermann

coveritytest, 2023-01-24:
And what about `<extract_partition part="2">sda2.img</extract_partition>`? Delete the line, too?

bgermann, 2023-01-24:
Yes, you get the idea.