Linutronix/elbe #363: Difference in /boot/grub between sda.img.tar.gz and rootfs.tar.xz

coveritytest, 2023-01-31:
I just discovered that the rfs partition of sda.img.tar.gz differs from the generated rootfs.tar.xz regarding the content of `/boot/grub`. In the rootfs.tar.xz there is only:

```
-rw-r--r-- 1 root root 4.7K Jan 30 14:56 grub.cfg
-rw-r--r-- 1 root root 2.3M Jan 30 14:55 unicode.pf2
```

but in the sda.img.tar.gz there is the whole GRUB setup:

```
drwxr-xr-x 2 root root    4096 Jan 30 15:56 fonts/
-rw-r--r-- 1 root root    5513 Jan 30 15:56 grub.cfg
-rw-r--r-- 1 root root    1024 Jan 30 15:56 grubenv
drwxr-xr-x 2 root root    4096 Jan 30 15:56 locale/
-rw-r--r-- 1 root root 2394108 Jan 30 15:55 unicode.pf2
drwxr-xr-x 2 root root   12288 Jan 30 15:56 x86_64-efi/
```

Why is that? As I don't want to use secure boot, I have the following GRUB/EFI packages installed:

```
grub-common
grub-efi-amd64
grub-efi-amd64-bin
grub2-common
libefiboot1
libefivar1
```

So this should be sufficient. I'm using this setup:

```
...
<images>
    <!-- use GPT for UEFI -->
    <gpthd>
        <name>sda.img</name>
        <size>16GiB</size>
        <!-- install grub -->
        <grub-install/>
        <!-- the EFI system partition -->
        <partition>
            <size>512MiB</size>
            <label>uefi</label>
            <bootable>true</bootable>
        </partition>
        <partition>
            <size>remain</size>
            <label>rfs</label>
        </partition>
    </gpthd>
</images>
<fstab>
    <bylabel>
        <label>rfs</label>
        <mountpoint>/</mountpoint>
        <fs>
            <type>ext4</type>
            <tune2fs>-i 0</tune2fs>
        </fs>
    </bylabel>
    <!-- the EFI system partition needs to be mounted at /boot/efi -->
    <bylabel>
        <label>uefi</label>
        <mountpoint>/boot/efi</mountpoint>
        <fs>
            <type>vfat</type>
        </fs>
    </bylabel>
</fstab>
...
<project-finetuning>
    <losetup img="sda.img">
        <extract_partition part="1">sda1.img</extract_partition>
        <command part="1">
            ls $ELBE_MNT
        </command>
    </losetup>
    <set_packer packer="tarxz">sda1.img</set_packer>
</project-finetuning>
...
```

bgermann, 2023-01-31:
This is because grbu-install is run on a loop-mounted device and the files from the installation with packages are copied to that (after the fact).

coveritytest, 2023-01-31:
Okay, thanks for the clarification. So in other words, with GRUB installations the rootfs.tar.xz cannot be used as it won't boot when copied to sda2.