Linutronix/elbe #368: debian-archive-keyring handling

yegorich, 2023-03-20:
As mentioned in #253, `debian-archive-keyring` installes the keys to `/usr/share/keyrings` but not to `/etc/apt/trusted.gpg.d`. What would be the proper way to handle this?

- mention this in the `README.adoc` for Ubuntu users
- handle this in the code via checking whether `debian-archive-*` are in `/usr/share/keyrings` or  `/etc/apt/trusted.gpg.d`

bgermann, 2023-03-20:
You should use the signed-by attribute of deb/deb-src lines as shown in https://elbe-rfs.org/download. Using trusted.gpg.d is bad because it allows global apt trust.

closed 2023-03-20 by bgermann

yegorich, 2023-03-20:
If I understand it correctly, `elbe-repo.pub.gpg` is required for elbe Debian package installation. What I need, are the keys for the `initvm`. On Ubuntu, I need the `debian-archive-keyring` package. The problem is that is installs the keys to `/usr/share/keyrings` but ELBE expects them in `/etc/apt/trusted.gpg.d`.

bgermann, 2023-03-20:
That is an inherant problem of not having Debian keys as trust anchor in Ubuntu systems. Please just copy them over; I cannot solve this.