Linutronix/elbe #370: libvirt.libvirtError: Cannot access storage file

yegorich, 2023-03-21:
On Ubuntu 22.04.1 LTS I get the following error when creating initvm:

`libvirt: Storage Driver error : Cannot access storage file '/home/user/elbe/initvm/initvm.img' (as uid:64055, gid:108): Permission denied`

I have fixed it via editing `/etc/libvirt/qemu.conf` and setting `user` to myself. This [article](https://ostechnix.com/solved-cannot-access-storage-file-permission-denied-error-in-kvm-libvirt/) provides background info.

Can this issue be fixed in elbe code?

bgermann, 2023-03-21:
@a-darwish Is this the bug that you reported to me some days ago?

bgermann, 2023-03-21:
It is already in the documentation that you have to add the elbe running user to the libvirt group.

bgermann, 2023-03-21:
The quickstart doc also says:
`adduser <youruser> libvirt-qemu`

yegorich, 2023-03-22:
No, this doesn't help. AFAIK, you don't have to add a user to `libvirt-qemu`. Only `kvm` and `libvirt` groups matter.

yegorich, 2023-03-22:
This issue seems to have something to do with the newer libvirt version (new security policy). See this [thread](https://bbs.archlinux.org/viewtopic.php?id=275126). 

I ended up adding read permissions to libvirt-qemu user to my home folder. Looks like specifying such permissions for initvm folder is not enough.

commit 99592e70 (2023-03-29): quickstart: add a user to kvm group instead of libvirt-qemu

Adding a user to the kvm group enables unprivileged access to the
/dev/kvm device, and libvirt group is enough to add unprivileged
access to the libvirt tools.

Fixes: #370

Fixes: b42c076d9e ("quickstart: describe how to use virtual machines as a regular user")
closed 2023-03-29 by bgermann