Linutronix/elbe #371: pbuilder: git: detected dubious ownership in repository

yegorich, 2023-03-22:
When trying to build own package via pbuilder, I get the following error:

```
   dh_clean
 dpkg-source -b .
dpkg-source: info: using source format '3.0 (git)'
fatal: detected dubious ownership in repository at '/var/cache/elbe/82654a86-1a66-4666-a271-2d53caf744d1/pdebuilder/current'
To add an exception for this directory, call:

        git config --global --add safe.directory /var/cache/elbe/82654a86-1a66-4666-a271-2d53caf744d1/pdebuilder/current
dpkg-source: error: git ls-files exited nonzero:
```
Some background info can be found in this [answer](https://stackoverflow.com/a/73100228/1113139).

Where can we add such a git config?

bgermann, 2023-03-22:
3.0 (git) is an experimental format and not supported by elbe pbuilder. Please use sbuild of other tools.

closed 2023-03-22 by bgermann

yegorich, 2023-03-23:
I know [git format](https://wiki.debian.org/GitSrc) is experimental. But I am using it for several years with `pbuilder` without any problem. This support is now broken because of the git policy changes.

I have made a dirty hack in my package's `debian/rules` file where I've just invoked `git config --global --add safe.directory '*'` in one of the initial stages. This workaround helped to produce a deb package with sources.

So adding something like this to pbuilder's initial setup, would restore the git format support:

`git config --global --add safe.directory /var/cache/elbe`

bgermann, 2023-03-23:
When is this happening? Is it the target's git version that is responsible or the initvm's?

bgermann, 2023-03-23:
If you want it fixed, please hand in patches.

yegorich, 2023-03-24:
When doing `elbe initvm attach` what password should I enter? According to the source code, it must be "foo". But if I use it, I get "Login incorrect" result.

I have tried to add user myself (`elbe control add_user ...`). I could see the newly created user via `elbe list_users`. But couldn't login as the new user either.

What am I doing wrong?