Linutronix/elbe #384: initvm fails with

tsipinakis, 2023-09-12:
Installed elbe version: `14.9+deb12+build1271`

We noticed that elbe `initvm` builds started failing in our CI infrastructure in the debian installer.

```
Sep 12 11:18:16 in-target: Setting up python3-elbe-buildenv (14.9.1+deb11+build1291) ...^M
Sep 12 11:18:17 in-target: Sorry: IndentationError: unindent does not match any outer indentation level (fetch_initvm_pkgs.py, line 179)^M
Sep 12 11:18:17 in-target: dpkg: error processing package python3-elbe-buildenv (--configure):^M
Sep 12 11:18:17 in-target:  installed python3-elbe-buildenv package post-installation script subprocess returned error exit status 1^M
Sep 12 11:18:17 in-target: dpkg: dependency problems prevent configuration of python3-elbe-daemon:^M
Sep 12 11:18:17 in-target:  python3-elbe-daemon depends on python3-elbe-buildenv (= 14.9.1+deb11+build1291); however:^M
Sep 12 11:18:17 in-target:   Package python3-elbe-buildenv is not configured yet.^M
Sep 12 11:18:17 in-target: ^M
Sep 12 11:18:17 in-target: dpkg: error processing package python3-elbe-daemon (--configure):^M
Sep 12 11:18:17 in-target:  dependency problems - leaving unconfigured^M
Sep 12 11:18:17 in-target: dpkg: dependency problems prevent configuration of python3-elbe-soap:^M
Sep 12 11:18:17 in-target:  python3-elbe-soap depends on python3-elbe-buildenv (= 14.9.1+deb11+build1291); however:^M
Sep 12 11:18:17 in-target:   Package python3-elbe-buildenv is not configured yet.^M
Sep 12 11:18:17 in-target:  python3-elbe-soap depends on python3-elbe-daemon (= 14.9.1+deb11+build1291); however:^M
Sep 12 11:18:17 in-target:   Package python3-elbe-daemon is not configured yet.^M
Sep 12 11:18:17 in-target: ^M
Sep 12 11:18:17 in-target: dpkg: error processing package python3-elbe-soap (--configure):^M
Sep 12 11:18:17 in-target:  dependency problems - leaving unconfigured^M
```

Looking a bit further it appears that release 14.9.1+deb11+build1291 which is pulled during the initvm contains some [invalid indentation in the try/except block](https://github.com/Linutronix/elbe/commit/fdc2e8623adc8c484c6e320f02b6699a764313c7#diff-8084ac88cfe236b457902faa711f3f64d26f09c724c5cc16b787092f2cc181bdR159-R179)

FedericoVaga, 2023-09-27:
Same problem here, and I think issue #387 is actually the same issue as this one

FedericoVaga, 2023-09-27:
Waiting for this to be fixed. How to instruct the initvm to use the previous ELBE release?

mdahlmanns, 2023-09-27:
Hi @FedericoVaga ! We had the same issue and circumvented it by cloning the repository, checking out a previous commit, and running `./elbe initvm create --devel` which instructs it to use the local code base instead of downloading it from GitHub.

Hope this helps!
Markus

FedericoVaga, 2023-09-27:
Indeed, thank you, @mdahlmanns, for the hint. I had forgotten about the `--devel` option.

FedericoVaga, 2023-09-29:
I did a quick patch here: https://github.com/FedericoVaga/elbe/tree/384-fix-indentation

 I will propose a pull-request later

FedericoVaga, 2023-10-06:
I see there is a new release including a fix for this indentation problem. A minor point, the `master` branch is still pointing to the `14.9.1` tag instead of the latest.

I think this issue can be closed, and https://github.com/Linutronix/elbe/issues/387 as well

ekrein89, 2023-10-25:
The Error was fixed as for version 14.9.2.
Closing the Issue

closed 2023-10-25 by ekrein89