Linutronix/elbe #386: SDK toolchain option

FedericoVaga, 2023-09-13:
Hi, I'm wondering if here we should also have `-B ` to tell `gcc` to look for other tools in the SDK and not on the host system.

https://github.com/Linutronix/elbe/blob/986d4a27f60608642bbeb9e13744e7ba633e30aa/elbepack/makofiles/environment-setup-elbe.mako#L58

I think that updating `PATH` is not enough, and that `gcc` internally builds a path to other tools. For example, I can see that `/usr/bin/ld` gets called from the host system. This should not happen

FedericoVaga, 2023-10-30:
An update on this issue. In our building process `PATH` was manipulated and consequentially we got tools invoked from the wrong environment.

So, the `environment-setup-elbe.mako` is ok, provided that `PATH` is untouched after sources. The `-B` options could be a safer way of telling GCC where to find the other tools.

t-8ch, 2024-04-29:
While `-B` would make it safer to find gcc-internal tools; in the face of a broken `PATH`, all other program lookups would still be unreliable.
So I'm not sure it makes much sense to add `-B`.

FedericoVaga, 2024-04-30:
I agree with you. Indeed, our problem was a manipulated `PATH`. Having had `-B` would have worked for us, but masking the fact of having a wrong `PATH`.

closed 2024-06-11 by t-8ch