Linutronix/elbe #387: Elbe installation fails.

murrellr, 2023-09-26:
I am trying to install and run ELBE on a VirtualBox virtual machine running Ubuntu.  After much effort I've got to the point where I'm trying to create the ELBE virtual machine and am getting an error.  I'm following the instructions at [ELBE Quickstart](https://elbe-rfs.org/docs/sphinx/article-quickstart.html#)  The host is Ubuntu "20.04.6 LTS (Focal Fossa)".  I execute the command `elbe initvm create` and get the following error:

Select and install software
---------------------------

!! ERROR: Installation step failed

An installation step failed. You can try to run the failing item again from the
menu, or skip it and choose something else. The failing step is: Select and 
install software

I'm a novice at Linux in general so I don't know where to look next.  How do I find out what failed?

murrellr, 2023-09-27:
After hacking around a little I see that it is the IndentationError:
`Sep 26 21:05:57 in-target: Sorry: IndentationError: unindent does not match any outer indentation level (fetch_initvm_pkgs.py, line...`

murrellr, 2023-09-28:
Is anyone going to address this?  I need to get ELBE installed so I can build our product's OS.

ekrein89, 2023-10-25:
The issue was fixed with the release of 14.9.2.
Closing now

closed 2023-10-25 by ekrein89