Linutronix/elbe #393: Fetch <pkg> from specific mirror

coveritytest, 2023-11-03:
If I have multiple mirrors defined via <binary></binary>, how can I tell a <pkg></pkg> to be fetched from a specific mirror? I did not find an example for that in the examples folder.

Is defining a version via <pkg version="x.y.z"> sufficient? And what if two mirrors have the same version present?

coveritytest, 2023-11-03:
Using `<pkg version="x.y.z">` does not work, it ignores the version and fetches a lower version from another mirror. Then it complains during build:

```
[INFO] Package List validation
[INFO] -----------------------
[INFO] 
[ERROR] Package 'abc' version '0.2-1' does not match installed version 0.1-1
```

jogness, 2023-11-03:
Please see the "pin" attribute of "pkg":

https://elbe-rfs.org/docs/sphinx/article-elbe-schema-reference.html#type-pkg

If two packages have the same version number, they better be identical.

coveritytest, 2023-11-07:
Not really working here. Do I have to pin the mirror itself, too? I set the package here:

`<pkg pin="0.16-1">package-xy</pkg>`

But it still installs the version 0.13-1 from the first mirror.

`2023-11-07T15:19:08.1871879Z elbe: 14.7
2023-11-07T15:19:08.1871958Z [INFO] 
2023-11-07T15:19:08.1872046Z [INFO] Apt Sources dump
2023-11-07T15:19:08.1872179Z [INFO] ----------------
2023-11-07T15:19:08.1872262Z [INFO] 
2023-11-07T15:19:08.1872456Z [INFO] deb [trusted=yes arch=amd64] http://ftp-stud.hs-esslingen.de/debian bookworm main
2023-11-07T15:19:08.1872686Z deb [trusted=yes] http://ftp.de.debian.org/debian bookworm non-free non-free-firmware
2023-11-07T15:19:08.1872930Z deb [trusted=yes] http://ftp.de.debian.org/debian-security bookworm-security main non-free non-free-firmware
2023-11-07T15:19:08.1873117Z deb [trusted=yes] https://xxx/debian bookworm main
2023-11-07T15:19:08.1874156Z [INFO] 
2023-11-07T15:19:08.1874237Z [INFO] 
2023-11-07T15:19:08.1874332Z [INFO] Apt Preferences dump
2023-11-07T15:19:08.1874464Z [INFO] --------------------
2023-11-07T15:19:08.1874557Z [INFO] 
2023-11-07T15:19:08.1874641Z [INFO] 
2023-11-07T15:19:08.1874722Z Package: package-xy
2023-11-07T15:19:08.1874822Z Pin: release n=0.16-1
2023-11-07T15:19:08.1874953Z Pin-Priority: 991
2023-11-07T15:19:08.1874995Z 
2023-11-07T15:19:08.1875025Z 
2023-11-07T15:19:08.1875133Z [INFO] 
2023-11-07T15:19:08.1875221Z [INFO] Installed Packages List
2023-11-07T15:19:08.1875367Z [INFO] -----------------------
2023-11-07T15:19:08.1904000Z [INFO] |package-xy|0.13-1|ftp-stud.hs-esslingen.de stable main`

What's am I missing?

jogness, 2023-11-07:
> Not really working here. Do I have to pin the mirror itself, too? I set the package here:
>
> `<pkg pin="0.16-1">package-xy</pkg>`
>
> But it still installs the version 0.13-1 from the first mirror. [...]

First, you never need to pin a higher version. Debian will automatically choose the highest version.

Are you sure your custom repository has the package in it? Is the package for the correct architecture? Is the package for the correct suite? Is it possible to install the package (i.e. are all of its dependencies also available)?

On your embedded system, you should be able to manually install your package with:

`$ apt install your-package`

I expect that also will not work. But it might provide some information about why.

coveritytest, 2023-11-16:
Issue can be closed, it finally worked.

closed 2023-11-16 by coveritytest