Linutronix/elbe #394: debugging a finetuning command

estergl, 2023-11-09:
i use as in the following example a finetuning command to fill a xml configuration file in my rootfs with some build informations. But the script was not executed during build process.
<finetuning>
<command>/usr/local/bin/myscript.sh -v 1.0.0 -d 2023-11-09 -n Productname</command>
</finetuning>

i can not see any trace information in elbe report or log file. Is there a standard mechanism to debug a finetuning command? I need to see some trace or debug messages during the execution of this command.

t-8ch, 2024-03-06:
The logs should don't show the command executed, but they should show the output of the command.

Example:
```xml
<command>echo hello world</command>
```
```
[CMD] chroot /var/cache/elbe/fe9af34f-af71-4a62-b1ea-3ff131d8cc9a/target /bin/sh
hello world
```

(This should be improved to also show the command)

t-8ch, 2024-04-29:
@estergl The command itself should be shown since cfe5e2242603758819d0b0d80d3fa40db0091591 .
Does it work for you?

closed 2024-06-11 by t-8ch