Linutronix/elbe #396: Cannot install – repository is empty

philippludwig, 2024-01-11:
The Quickstart guide tells us to use the repository @ http://debian.linutronix.de/elbe/, but the server only yields a 403 error.

t-8ch, 2024-02-05:
How did you access the repository?
With a browser directly you will receive a 403.

This URL is meant to be used to configure apt which will use it as base to construct other URLs which do work.
See for example http://debian.linutronix.de/elbe/dists/bookworm/Release

philippludwig, 2024-02-05:
I see. Adding `dists/bookworm/Release` did not occur to me.

Other debian mirrors, e.g. http://ftp.be.debian.org/debian/, have a working page on the base URL. So I just assumed that the repository is broken.

closed 2024-02-05 by philippludwig