Linutronix/elbe #408: elbe v15.0: finetuning/raw_cmd unintentionally prepends "/bin/sh" to the command

dermaulwurfpauli, 2024-07-10:
Building a diet image (BusyBox, no bash/dash package installed) with elbe v15.0 causes the build to fail when using a raw_cmd in the finetuning section.
Elbe v15.0 prepends "/bin/sh" to the raw_cmd. This symlink is not available until, in this case, '/bin/busybox --install -s' was executed.
Has worked so far correctly with elbe v14.9.3

Log elbe v15.0:
`[CMD] /usr/sbin/chroot /var/cache/elbe/019096e3-2cc0-77aa-aa73-bf3ef5abda09/target /bin/sh -c '/bin/busybox --install -s'
 /usr/sbin/chroot: failed to run command '/bin/sh': No such file or directory
 [ERROR]Finetuning Error: Command '['/usr/sbin/chroot', '/var/cache/elbe/019096e3-2cc0-77aa-aa73-bf3ef5abda09/target', '/bin/sh', '-c', '/bin/busybox --install -s']' returned non-zero exit status 127.
 Traceback (most recent call last):
 File "/usr/lib/python3/dist-packages/elbepack/finetuning.py", line 627, in do_finetuning
 action.execute(buildenv, target)
 File "/usr/lib/python3/dist-packages/elbepack/finetuning.py", line 303, in execute
 chroot(target.path, self.node.et.text)
 File "/usr/lib/python3/dist-packages/elbepack/shellhelper.py", line 162, in chroot
 do(['/usr/sbin/chroot', directory, '/bin/sh', '-c', cmd], env_add=new_env, **kwargs)
 File "/usr/lib/python3/dist-packages/elbepack/shellhelper.py", line 134, in do
 run(cmd, shell=_is_shell_cmd(cmd), env=new_env, stdout=ELBE_LOGGING, stderr=subprocess.STDOUT,
 File "/usr/lib/python3/dist-packages/elbepack/shellhelper.py", line 95, in run
 return subprocess.run(cmd, stdout=stdout, stderr=stderr, check=check, **kwargs)
 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 File "/usr/lib/python3.11/subprocess.py", line 571, in run
 raise CalledProcessError(retcode, process.args,
 subprocess.CalledProcessError: Command '['/usr/sbin/chroot', '/var/cache/elbe/019096e3-2cc0-77aa-aa73-bf3ef5abda09/target', '/bin/sh', '-c', '/bin/busybox --install -s']' returned non-zero exit status 127.`

Log elbe v14.9.3:
`[CMD] chroot /var/cache/elbe/df2bc404-dce2-4000-ae7a-1e88ccad14ae/target /bin/busybox --install -s`

t-8ch, 2024-07-11:
We should probably execute `<raw_cmd>` without a shell.
If somebody needs a shell they can use `<command>`.

The problem is that historically `<raw_cmd>` also used a shell, but the one from the build environment.
So something like `<raw_cmd>cat /etc/hostname > /etc/hostname2</raw_cmd>` would effectively copy the hostname file from the *target* system to the *build* environment, which "works" but is clearly wrong.

t-8ch, 2024-07-11:
This should be fixed by the changes in https://lists.linutronix.de/pipermail/elbe-devel/2024-July/007179.html .
(Only the single change to `elbepack/finetuning.py` should be enough to test)

commit 6b90d50c (2024-07-12): elbepack: finetuning: run raw_cmd without a shell

In diet mode there very well may not be a shell available in the target
system during finetuning.
Run `<raw_cmd>` actions without a shell.
Historically a shell was used for `<raw_cmd>`, but that was the buildenv
one, leading to surprising results.

Removing the shell from `<raw_cmd>` now also makes the difference
between it and `<command>` clear.

Also align the schema documentation and tests to that effect.

Closes #408
closed 2024-07-12 by t-8ch