Linutronix/elbe #409: Elbe failed to run the hello world example from tutorial on a fresh installed debian

cnnblike, 2024-07-26:
Hi, I'm now running a fresh installed debian 12.6 with latest elbe from your official repository.
However it keep failing with strictly following the tutorial on <https://elbe-rfs.org/docs/sphinx/v15.0/article-quickstart.html> on the step "Submitting an XML file".

Minor tweaking around SubmitAction(InitVMAction) by adjusting the following line:
```python
ps = run_elbe(['initvm', 'ensure', *additional_params], capture_output=False,
```
Gives me some more information:
```
cnnblike@kl-desktop:~/workplace/elbe$ elbe initvm submit examples/x86_64-pc-rescue-busybox-dyn-cpio.xml
*
***********Waited for 2.0 minutes and the daemon is still not active.
Starting the initvm Failed
Giving up
```

I guess this shall be related to something going wrong on the previous "setting up initvm" step.
I've attached log output of that step, could you please take a look into it?

[installer.log](https://github.com/user-attachments/files/16395666/installer.log)

t-8ch, 2024-07-29:
The `installer.log` you sent looks fine.

What is the output of `virsh --connect qemu:///system dominfo initvm`?

cnnblike, 2024-07-31:
Hi, t-8ch@, I've erase the full image on my machine, another fresh install stuck in the first step with log at `/var/log/syslog` ends like below, it seems like some magic source from <http://ftp.de.debian.org/debian/pr> is refusing my connection. 
```
Jul 30 17:27:57 in-target: Get:179 http://ftp.de.debian.org/debian bullseye-backports/main amd64 reprepro amd64 5.3.1-1~bpo11+1 [453 kB]
Jul 30 17:27:57 in-target: Get:180 http://ftp.de.debian.org/debian bullseye/main amd64 squashfs-tools amd64 1:4.4-2+deb11u2 [135 kB]
Jul 30 17:27:57 in-target: Get:181 http://ftp.de.debian.org/debian bullseye/main amd64 sudo amd64 1.9.5p2-3+deb11u1 [1,061 kB]
Jul 30 17:27:58 in-target: Get:182 http://ftp.de.debian.org/debian bullseye/main amd64 subversion amd64 1.14.1-3+deb11u1 [997 kB]
Jul 30 17:28:00 in-target: Get:183 http://ftp.de.debian.org/debian bullseye/main amd64 python3-tz all 2021.1-1 [34.8 kB]
Jul 30 17:28:01 in-target: Get:184 http://ftp.de.debian.org/debian bullseye/main amd64 python3-pyparsing all 2.4.7-1 [109 kB]
Jul 30 17:28:01 in-target: Get:185 http://ftp.de.debian.org/debian bullseye/main amd64 python3-spyne all 2.13.16-1 [278 kB]
Jul 30 17:28:03 in-target: Fetched 118 MB in 7min 27s (264 kB/s)
Jul 30 17:28:03 in-target: E: Failed to fetch http://ftp.de.debian.org/debian/pr. Remote end closed connection [IP: 141.76.2.4 80]deb  Error reading from r
Jul 30 17:28:03 in-target: E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
Jul 30 17:28:03 main-menu[2749]: (process:26786): dpkg-divert: warning: divertirous, use --no-renamestop-daemon' from an Essential package with rename is r
Jul 30 17:28:03 main-menu[2749]: (process:26786): dpkg-divert: warning: divertirous, use --no-renamestop-daemon' from an Essential package with rename is r
Jul 30 17:28:03 main-menu[2749]: (process:26786): dpkg-divert: warning: divertirous, use --no-renamestop-daemon' from an Essential package with rename is r
Jul 30 17:28:03 main-menu[2749]: (process:26786): dpkg-divert: warning: divertirous, use --no-renamestop-daemon' from an Essential package with rename is r
Jul 30 17:28:03 main-menu[2749]: (process:26786): dpkg-divert: warning: divertirous, use --no-renamestop-daemon' from an Essential package with rename is r
Jul 30 17:28:03 main-menu[2749]: WARNING **: Configuring 'pkgsel' failed with error code 100
Jul 30 17:28:03 main-menu[2749]: WARNING **: Menu item 'pkgsel' failed.
Jul 30 17:32:22 main-menu[2749]: INFO: Modifying debconf priority limit from 'high' to 'medium'
Jul 30 17:32:22 debconf: Setting debconf/priority to medium
Jul 30 17:32:23 main-menu[2749]: INFO: Falling back to the package description for brltty-udeb
Jul 30 17:32:29 main-menu[2749]: INFO: Falling back to the package description for brltty-udeb
Jul 30 17:32:29 main-menu[2749]: INFO: Menu item 'di-utils-shell' selected
```

after re-enable from the installation sequence, it's now like this:
```
20: Execute a shell,    
 21: Abort the installation,    
Prompt: '?' for help, default=14> 

Select and install software  ... 1%... 10%
Select and install software
---------------------------

Applying updates on a frequent basis is an important part of keeping the system
secure.

By default, security updates are not automatically installed, as security 
advisories should be reviewed before manual installation of the updates using 
standard package management tools.

Alternatively the unattended-upgrades package can be installed, which will 
install security updates automatically. Note however that automatic 
installation of updates may occasionally cause unexpected downtime of services 
provided by this machine in the rare cases where the update is not fully 
backward-compatible, or where the security advisory requires the administrator 
to perform some other manual operation.
The system is ghaveged: haveged: Stopping due to signal 15                      71%... 82%
Installing GRUB boot loader  ... 16%... 3... 21%... 32%... 42%... 50%... 60%... 
Sent SIGKILL to all processes.. 3%... 10%
Requesting system reboot
[17083.464278] reboot: Restarting system
qemu-img create -f qcow2 -F qcow2 -b initvm-base.img initvm.img
Formatting 'initvm.img', fmt=qcow2 cluster_size=65536 extended_l2=off compression_type=zlib size=85899345920 backing_file=initvm-base.img backing_fmt=qcow2 lazy_refcounts=of6
libvirt: Storage Driver error : Cannot access storage file '/home/cnnblike/workplace/elbe-base/initvm/initvm.img' (as uid:64055, gid:64055): Permission denied
Traceback (most recent call last):
  File "/usr/bin/elbe", line 56, in <module>
    cmdmod.run_command(sys.argv[2:])
  File "/usr/lib/python3/dist-packages/elbepack/commands/initvm.py", line 94, in run_command
    action.execute(directory, opt, args[1:])
  File "/usr/lib/python3/dist-packages/elbepack/initvmaction.py", line 164, in execute
    self.initvm.create()
  File "/usr/lib/python3/dist-packages/libvirt.py", line 1373, in create
    raise libvirtError('virDomainCreate() failed')
libvirt.libvirtError: Cannot access storage file '/home/cnnblike/workplace/elbe-base/initvm/initvm.img' (as uid:64055, gid:64055): Permission denied
Starting the initvm Failed
Giving up
```

t-8ch, 2024-07-31:
I'm a bit confused be the two logs.

The presence of `installer.log` in the first one indicates v15.0 while the usage of `ftp.de.debian.org bullseye` and the permission error indicate an older version.

Could you clarify which version of elbe you are using and try to get back to v15.0?

cnnblike, 2024-07-31:
hi, both of them are on fresh installed debian 12 (running on different flash drive but I don't think it's the root cause).  
The second one log was from this:
```
cnnblike@kl-desktop:~/workplace/elbe-base$ elbe --version
elbe v14.9.3
cnnblike@kl-desktop:~/workplace/elbe-base$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
```
Maybe it's caused by different debian apt source list? would you mind share which mirror do you use?

t-8ch, 2024-07-31:
The official mirror is at `http://debian.linutronix.de`.
See: https://elbe-rfs.org/download.html

This should give the following source line:
```
deb [signed-by=/usr/share/keyrings/elbe-archive-keyring.gpg] http://debian.linutronix.de/elbe bookworm main
```

Could you share the output of `apt policy elbe`?

You can also enable the elbe repo through `extrepo`:

```bash
extrepo enable elbe
apt update
apt install elbe
```

cnnblike, 2024-07-31:
Aahh, I think I spot the blind point.
I just copied from the source setting up line from tutorial and use the "buster" instead of "bookworm" here without much think about this.
I'm now setting up with elbe 15.0, let me check if things goes correct.

t-8ch, 2024-07-31:
Does `curl localhost:7587/soap?wsdl` return output?

t-8ch, 2024-07-31:
This is only supposed to work outside the initvm.
What about `curl localhost:7587/?wsdl`

cnnblike, 2024-07-31:
also not found, 
I just checked occupied port
```
cnnblike@kl-desktop:~/workplace/elbe-base/elbe$ sudo netstat -tunlp
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 127.0.0.1:631           0.0.0.0:*               LISTEN      900/cupsd           
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      921/sshd: /usr/sbin 
tcp        0      0 0.0.0.0:7587            0.0.0.0:*               LISTEN      13266/qemu-system-x 
```
the request should have been routed to qemu machine from my understanding.

t-8ch, 2024-07-31:
Can you apply c105bf6fc42a1369cb621b3ebeded040578e9237 inside the initvm and then `systemctl restart python3-elbe-daemon`.
I'm wondering why this now(?) also affects the soap server, will investigate.

t-8ch, 2024-07-31:
FYI for the specific example you are using you also need to apply 6b90d50cfdcdf3414bf162dc4c1c80390665d533
(at least the one line to `elbepack/finetuning.py`)

I'll try to get a release out with these fixes as fast as possible.

cnnblike, 2024-07-31:
em, maybe that's why, the elbepacks inside that vm is the old version I guess
```python
import elbepack.daemons

def _not_found(start_response):
    start_response('404 Not Found', [('Content-Type', 'text/plain')])
    return [b'not found']

class _WsgiDispatcher:
    def __init__(self, mapping):
        self.mapping = mapping

    def __call__(self, environ, start_response):
        path_info = environ['PATH_INFO']
        parts = path_info.split('/', maxsplit=2)
        if len(parts) != 3:
            return _not_found(start_response)
        _, app_name, _ = parts

        app = self.mapping.get(app_name)
```

cnnblike, 2024-07-31:
ok, I solved the issue by directly copy whatever we have under in the master branch fro elbepack. 
It looks like the latest one should be good now.
```
cnnblike@kl-desktop:~/workplace/elbe-base/elbe$ curl localhost:7587/soap?wsdl
<?xml version='1.0' encoding='UTF-8'?>
<wsdl:definitions xmlns:xs="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-
instance" xmlns:plink="http://schemas.xmlsoap.org/ws/2003/05/partner-link/" xmlns:wsdlsoap11="http://schem
```

cnnblike, 2024-07-31:
but it seems like it still can't build correctly if I submit  xml file to the endpoint:
```
<pre><span style="color:#26A269"><b>cnnblike@kl-desktop</b></span>:<span style="color:#12488B"><b>~/workplace/elbe-base/elbe</b></span>$ elbe initvm submit examples/x86_64-pc-rescue-busybox-dyn-cpio.xml
Starting the initvm Failed
Giving up
</pre>
```

t-8ch, 2024-08-01:
I'll try to reproduce this.

But if you have a git checkout anyways, you can use elbe directly from there.
```
./elbe initvm destroy
./elbe initvm create
./elbe initvm submit examples/x86_64-pc-rescue-busybox-dyn-cpio.xml
```

cnnblike, 2024-08-01:
Hi, sorry for the loong troubleshooting thread, but it still don't work on my side. 
[output.log](https://github.com/user-attachments/files/16458863/output.log)

t-8ch, 2024-08-02:
Please do a git pull.
The file `examples/x86_64-pc-rescue-busybox-dyn-cpio.xml` you are using was temporarily in another location.
(See 149efadc6da6dbf3b26acc2979f24e173a7bbf2c)

The second error on `initvm destroy` is also fixed.

bookworm should be fine to use.

t-8ch, 2024-08-02:
You're welcome. I hope to release a fixed version soon.